SanPen/VeraGrid #143: Unbalanced short circuit + "Tiny" in helm_power_flow.py

poypoyan, 2021-12-30:
Hi! Thanks for this software. I'll start understanding it.

1) I want to code some unbalanced short circuit analyses. Yeah this is not new stuff, but I want to try contributing to open source :D.

2) In `helm_power_flow.py`, `epsilon` function, I think `Tiny` should be `np.finfo(complex).tiny`, because `min = -max`.

SanPen, 2021-12-30:
Hi!

Unbalance short circuit would be a very nice addition indeed. I believe you'll need some parameters that are currently missing like r0, x0, etc... Let me know and I'll add them. In the mean time, you may work in a simple standalone funtion that takes those as input.

About `Tiny` you're probably right. I changed it already.

poypoyan, 2021-12-30:
Yes I'll make standalone functions for now, then I'll reply.

Where can I find recent changes? Thanks!

SanPen, 2021-12-30:
The most updated changes and ongoing work is on the devel branch.

poypoyan, 2022-02-12:
Hello! I now have a "sketch" function for unbalanced short-circuit analysis. This resembles the `short_circuit_3p` function in ShortCircuitStudies. This is very preliminary: I have not yet verified many cases (see the WARNING comment), and I still need to find software to compare to. Also, the size of ndarrays are different because each phase have different values of current, voltage, etc.

I have some questions:

1. [Here](https://github.com/SanPen/GridCal/blob/master/src/GridCal/Engine/Simulations/ShortCircuitStudies/short_circuit.py#L30): what's the purpose of `len(bus_idx)`?
2. Are the outputs of Short circuit analysis used elsewhere in GridCal? I don't think so, but I am not experienced enough in EE to know.

SanPen, 2022-02-13:
Hi!

1) It was something I came up with that doesn't make much sense. It is meant to average the voltage increment by the number of failed buses, which is not how it should work. In practice it works because I do use multiple failures. So skip it.
I still haven't seen a multi-failure calculation method explained. If know about one, please let me know.

2) No they're not. The results are displayed in the GUI but that's it. Do you thing you'll need them elsewhere?

About the later integration of your code:

- I would not use a 3D Zbus. This is something I thought about when designing my first three phase software (not gridcal...). The right thing to have is a matrix that is 2D but with 3 times more columns and rows. Have a look at [this](https://github.com/SanPen/GridCal/blob/master/src/research/three_phase/Engine/circuit_matrix_inputs.py). However I believe we can live with 3 separated NxN matrices Z0, Z1 and Z2. All this is because of computational efficiency later.

- One big issue, at least in my mind, is the proper generation of Z0, Z1 and Z2. What do you have in mind here? I am pretty sure we'll need to add more data to the objects to know things like the transformer connection (delta-star, star-star, etc.)

Best regards,
Santiago

poypoyan, 2022-02-13:
> One big issue, at least in my mind, is the proper generation of Z0, Z1 and Z2...

Yeah I also realized this. Transformer connection is needed for Z0. Adding more data looks like a significant change for me for now.

> I would not use a 3D Zbus...

I'll edit my code for this.

I would not add unbalanced short circuit for now. I'll close this issue.

poypoyan, 2022-04-05:
This is about multiple balanced faults. I am still finding sources about it, but I have an idea. Assuming that Zf's are in wye already, one requirement is that at post-fault, the voltages of faulted buses must simply be Zf*If. So I thought that the following equation gives the correct way to compute the fault currents:

V_fbuses = Z_submatrix * I_fbuses + Z_fbuses * I_fbuses

Where  {V,Z,I}_fbuses are vectors, V_fbuses are prefault, and Z_submatrix only includes Z values involving the faulted buses. Then for faulted buses, V' = V - Z * I = Zf*If. (Just adjust the signs)

I need to think of another requirement (something about "linearity"?) to guarantee the above equation. By the way, I'll implement this in `short_circuit.py`.

SanPen, 2022-04-06:
Hi, 

Could you please indicate the documentation that you're following? in that way I'll be able to comment :D

Also, in the code fork, it would be nice to have the reference in the docstrings for the future.

BR,
Santiago

poypoyan, 2022-04-06:
I honestly just came up with this. Most literature about simultaneous faults are more general, involving unbalanced faults and hence, uses different modelling such that extracting for the balanced faults only is cumbersome (for me). As for my 2 requirements in docs, I am sure about the second one, but the first one about linear relationship needs proof/justification.

poypoyan, 2022-04-09:
Thanks! I'll work on (multiple) unbalanced short circuit if I have time to formulate and find references.

SanPen, 2022-04-09:
Thanks to you for the contribution. If you let me know your name I'll add it to the credits as contributor.

poypoyan, 2022-04-09:
I now added my real name on this Github profile. Hope I can contribute more aside from this issue.

poypoyan, 2022-05-01:
Regarding multiple unbalanced short circuit, I am convinced we ditch symmetric components, and instead do it more directly. The following are the resources I found that closely resembles my "matrix" approach for multiple balanced:
https://www.sciencedirect.com/science/article/abs/pii/S014206159600052X
https://ieeexplore.ieee.org/document/500821

These are paywalled. I can provide the PDFs by email maybe.

SanPen, 2022-05-22:
Hi!,

Sorry for the delay on my side. Indeed the full blown 3-phase approach is the best and the "simplest" since the device modes do not change but are expanded. Both articles are fine, you may also read if you find it interesting the book "[Distribution system modelling](https://www.amazon.com/Distribution-Modeling-Analysis-William-Kersting/dp/1498772137)" I think it includes the best 3-phase modelling explanations.

Do you have any suggestion on how to continue?

BR,
Santiago

SanPen, 2022-06-16:
Hi!

No rush of any kind, when you can work on it it'll be great.

I must admit I do not have a lot of experience with short circuits either. But, for reliability studies it is quite common to consider the failure of "critical" pairs or triplets of lines at the same time (no N-2 or N-3 strictly). So at the time I thought that there should be a way to implement that for short circuits too. Maybe using superposition but I did not investigate much.

All the mainstream transmission software uses symmetrical components for the short circuit. Chapter 5 of Computer Analysis of Power Systems (one of the best books on power systems ever written) deals with balanced and unbalance faults and the whole book resembles to the paper you just cited. I would follow this book since it provides algorithms and examples.

BR,
Santiago

poypoyan, 2022-07-08:
Update: prototype is done. 😅  This is the direct.py [here](https://gist.github.com/poypoyan/1d68424596cbe9fae01b644840b27c96) (I linked this gist so many times here).

Now I did not completely follow the papers above. The inputs are still symmetrical components, but instead of directly solving using those, those are 'brought back' to phases ABC, and the full 3-phase Ybus matrix is built. Then inversing it gives the 3-phase Zbus matrix. This Zbus matrix is then manipulated for fault current calculations.

Four fault types, all short-circuit, are provided: Single L-G, Double L-G, L-L, and (Wye) Balanced. I derived the calculations for each based on this equation: $V_{post} = V_{pre} + Z I_f \implies V_{pre} = V_{post} - Z I_f$ (I'll start the documentation later).

Now for parameters, I am not knowledgeable on this, but my reference[1] says the following:
* For components that do not move, like transformers and transmission lines, Z1=Z2 *always*. 
* For transmission lines, Z0 is "quite involved", involving "Carson's Formula". Too complex for me right now.
* For transmission lines, zero-sequence circuit depends on transformer connection. I implemented 6 basic ones: Yg-Yg, Yg-Y, Yg-D, Y-Y, Y-D, D-D; the zero-sequence circuits are everywhere on the Internet. For these connections, fortunately $Z_0 = Z_1$. My comment: extra impedances may be present for connections I didn't implement (open-delta, auto-transformers, Wye with neutral-ground impedance, etc.). So I provided  `extraZ` in `add_tr` function for those.
* For (wye-connected) generators/loads, if there is neutral-ground impedance $Z_n$, then $3Z_n$ is added to $Z_0$.

So for now, I request adding these parameters for now: Z0 (or R0, X0) for transmission line, and Z2 and Z0 for generators and loads. I see that there are other components in GridCal. No idea for fault analysis for those.

Let me know what to do next 😃.

[1] "Power System Analysis" (3rd edition) by Saadat.

SanPen, 2022-07-08:
Hi!

Great work. Quickly looking at the gist, there are some things there need to evolve.

First, I'm 100% agree with creating the 3-phase array from the sequence ones. Only that we need to do that efficiently. [Here](https://github.com/SanPen/GridCal/blob/master/src/GridCal/Engine/Core/admittance_matrices.py) you'll see how the admittance matrices are computed. In GridCal all the branches are a subset of the [FUBM](https://modeladopractico.readthedocs.io/es/latest/fubm_acdc_model.html) (Full Unified Branch Model) plus some virtual transformers that occur when a device has different nominal voltage at the terminals than the nominal voltages of the buses. In short we need to adapt the academic model to the industrial one. To be honest, I'm not sure how to do it from sequence components. Maybe we need to create a 3-phase FUBM model first and from there the admittance calculation is the same.

GridCal wise, what we should do is to create a branch and make all the modifications there: 
- Add R0 and X0 to the branches and generators
- Add the winding type to the transformer
- Modify the data compiler 
- Serve the data to short circuit methods. 
- Start with the specific routines

About getting super detailed 3-phase matrices from the lines is already there.

Let me know.
BR,
Santiago

poypoyan, 2022-07-28:
Alright! I am now quite confident about the phase shifts, especially because of what I did is the same to matrix 3.4.13 of Arrillaga, except I hard-coded the simplifications $y'_m = y''_m = y'''_m = 0$, $y_p \approx y_m \approx y_s$ described in Arrillaga, which is actually also the simplifications found in the [this article](https://ieeexplore.ieee.org/abstract/document/1490591) brought by @JosepFanals.

However, I also take into account that the bus voltage at delta side can lead **or** lag by 30 degrees (this is related to 'DAB' vs 'DAC' connection and to vector group notation like Dyn1 vs Dyn11, something like that). In my code comment [here](https://gist.github.com/poypoyan/1d68424596cbe9fae01b644840b27c96#file-3pzbus-py-L62), I only used 'dlead' and 'dlag' for specifying lead and lag phase shift, resp., of bus voltage at delta side. May be this is nitpicking, because in the wye-side, post-fault bus voltage and fault current is *the same* regardless of lead or lag or even when removing the phase shifts! But on the delta side, there is difference in post-fault.

Because of that, the placing of $\angle{30\degree}$ and  $\angle{-30\degree}$ may swap (in my code, this is [lines 70-80](https://gist.github.com/poypoyan/1d68424596cbe9fae01b644840b27c96#file-3pzbus-py-L70)). Now which is which for dlead and dlag, it needs explanation (I have some ideas but that's for later). What I did for now (the "mini-tests") is to see the pre-fault injected currents (in my code, this is [line 190](https://gist.github.com/poypoyan/1d68424596cbe9fae01b644840b27c96#file-3pzbus-py-L190)): in pre-fault, injected currents should be all zero for buses without generators or loads connecting. Tell me if you want me to remove this consideration.

Moving on, I don't think we need to implement the whole algorithm in the above article simply because the pre-fault FUBM is balanced, hence we can compute for negative and zero sequence stuff later (i.e. just when needed in fault analysis).

Now for fault current computation, there are three directions to go:
1) Build 3N X 3N phase coordinate Ybus matrix by converting back the sequence components. **Pros:** no matrix inversion of large matrix. **Cons:** fault current calculation is slower than the other approaches, because the resulting linear system solves the post-fault bus voltages first, and involves almost the whole Ybus matrix. Fault current computation should be the priority, and in commercial softwares afaik, fault currents for each buses are computed.
2) My approach: Build the 3N X 3N phase coordinate Zbus matrix by converting back the sequence components. **Pros**: more general faults (e.g. line-to-ground fault in Phase B). Not sure about open circuit faults though. **Cons**: slower, because inversing 3N x 3N Ybus to Zbus takes long time.
3) The more "standard" approach: 3 separate NxN Zbus matrices (for 0, +, - sequence). No "converting back" anymore. **Pros: ** faster and parallelizable. **Cons:** Afaik, only computes single bus faults, and on specific phases only (L-G on phase A, L-L and L-L-G on phase B and C; I can think of work-around for this though). I think most of the time, single bus faults are the most useful.

Now what I do is to also implement the 3rd approach, and all of this into another github repo, instead of just a gist.

Best,
Ameer

JosepFanals, 2022-08-04:
Hi all,

I think I have managed to get it working in the sequences. You can find the code [here](https://gist.github.com/JosepFanals/34ca29bd5fd420f1f12535aa1b40122d). It basically follows the same structure as @poypoyan's code, with the difference that there is no going back to ABC. As far as I can tell it is relatively straightforward to follow. The results match with the ones of the example 10.6b found in the book Hadi Saadat - Power System Analysis. 

I agree on the fact that it is no big deal if the program can only compute conventional single bus faults. If faults take place in another phase (for instance a LG fault appears in phase B instead of A), there are ways to consider this. Basically we would have to add a shift in the fault current. Besides, if we wanted to compute various simultaneous faults, the application of the superposition principle should still be valid. In my humble opinion, it is better to opt for the 3rd option @poypoyan mentions as it seems to be the most efficient.

When it comes to the modelling, Z0 impedances have to be given at the end of the day. This had me thinking because in the perfectly balanced typical power flow we employ representative phase impedances for the lines, which already include the self impedance $\underline{Z}_s$ and the mutual impedance $\underline{Z}_m$, being $\underline{Z}_1=\underline{Z}_2=\underline{Z}_s - \underline{Z}_m$. However, $\underline{Z}_0=\underline{Z}_s+2\underline{Z}_m$. I am saying all this because in the FUBM we do not know the values of $\underline{Z}_s$ and $\underline{Z}_m$ separetely. Thus, $\underline{Z}_0$ should be specified by the user somehow. So while I feel like we can adapt the FUBM for positive and negative sequences, I do not know if we can use it systematically for the zero sequence. @SanPen, any thoughts on how to include the FUBM?

Overall I believe we are now at a point where we should start thinking how to modify GridCal as such. I can collaborate on that if needed.

SanPen, 2022-08-05:
Hi, 

Having a look at the codes, there's still some refactoring to be done in order to plug it into GridCal:

- No need to compose Ybus and then split it into Y0, Y1 and Y2. Better to compose those matrices separately using the `compute_admittances` [function](https://github.com/SanPen/GridCal/blob/master/src/GridCal/Engine/Core/admittance_matrices.py).
- For that, the functions `add_line`, `add_transformer` etc. need to be at least modified to represent the FUMB components in sequences. This is as simple as having (R0, X0, G0, B0) (R1, X1, G1, B1) (R2, X2, G2, B2) in all the devices. Something that's almost there in the devel branch. That needs to be reviewed to make sense. 

The rest should be straight forward.

poypoyan, 2022-08-05:
@JosepFanals Many thanks for working on this. Of course, I am fine with the 3rd option. I know that documentation is outdated, but [this](https://gridcal.readthedocs.io/en/latest/theory/branch_model.html) and the two papers referenced [here](https://modeladopractico.readthedocs.io/es/latest/fubm_acdc_model.html) greatly helped me understand the `compute_admittances` function.

I am not sure if I can still contribute in the Gridcal code itself, but I'll still think about zero sequence and about the `add_line` and `add_transformer`. I let you know if I came up with something.

Best,
Ameer

JosepFanals, 2022-08-10:
Hi all,

First of all, I have polished the calculation of balanced faults. This includes adding the R1 and X1 values of generators and batteries to the admittance matrix. Then, the typical approach where $V = V_{pre} + \Delta V$ is followed. I have reformulated the short-circuit power calculation to $SCC_i = V_i^2 / Z_{i,i}$, which is the standard way of defining it. This can be found in [this fork](https://github.com/SanPen/GridCal/compare/devel...JosepFanals:GridCal:devel?expand=1). 

My idea is to tackle the unbalances tomorrow. The plan is as follows. The admittance matrices Y0, Y1 and Y2 will integrate the corresponding generator and battery impedances as shunt elements (they just affect the diagonal terms, nothing new). For the FUBM used in branch elements such as lines and transformers, my idea is to follow the `compute_admittances` function you mentioned. From my perspective, $G_{sw}$ and $B_{eq}$ should only be present in positive sequence because they represent the switching power losses and the absorption of reactive power. Being these magnitudes powers originally used for balanced conditions, I understand it is safe to ignore them in the negative and zero sequence. Then we will get to a reduced model that includes the ideal transformer and the $\pi$ branch model. For this it is just a matter of using the expressions from line 73 to 81 in the `compute_admittances` function. The only tricky part is the transformer connections. As far as I can tell this information will have to be added to the transformer device.  Besides, it has to be seen how to include the phase shifts in the off-diagonal terms, and also, some Z0 values will have to go to infinity depending on the connection. I will try my best to not rely on many conditionals, not sure how simple it will be.

@poypoyan, thank you very much for the references. I still think it would be nice if you could get the faults working in abc since it would be useful to compare speeds and double-check the results. Also, I believe it is necessary to have a function that starting from the slack bus, adds the successive +-30 degrees phase shifts of transformers to the prefault voltages. I am quite sure some search algorithm is needed for that, although I am no expert in this regard. Do you have any suggestions on how to approach it?

SanPen, 2022-08-11:
Hi Josep, 

I started a review for a pull request from your branch. There are some minor things to update.

If you need me to help you change things, please let me know.

Best regards,
Santiago

JosepFanals, 2022-08-11:
Hi Santiago,

Thank you very much for your reviews. I think I have gone all over them as of now, but there are probably many aspects to improve in terms of readability. Feel free to modify whatever you find convenient. If something is not clear, I will try my best to clarify it. 

In terms of functionality, I believe the code does what was intended. However, we still need some sort of algorithm to initialize the positive sequence voltages adding the phase shifts. Some search path could work but perhaps it is not the smartest way. I will keep thinking about this. If you have a more solid idea on how to approach it, please let me know. 

Best regards,
Josep Fanals

SanPen, 2022-08-16:
Hi Guys, 

The unbalanced short circuit is finally integrated in the devel branch and will be available in the 4.6.0 release.

Thank you very much to both of you!

best regards,
Santiago

closed 2022-08-24 by poypoyan