SanPen/VeraGrid #146: [FEATURE REQUEST] DC Power Flow (or even optimal power flow)

yasirroni, 2022-03-23:
After digging the docs, I can't find anything related to DC systems. I would like to be able to run HVDC and Microgrid DC systems. Hybrid capability with AC systems would be even better.

Anyway, a great work!

SanPen, 2022-03-24:
Hi Muhammad!

Have you tried to set the buses to dc mode when drawing?

In principle GridCal handles AC/DC multiterminal. Let me know if you manage.

Best regards,
Santiago

yasirroni, 2022-03-24:
Is there any docs or example to use that feature (because I can't find any)? Thank you.

SanPen, 2022-03-24:
not really, we're behind with the docs.
It should be very intuitive, try it and let me know if you manage.

br, 
Santiago

yasirroni, 2022-03-28:
Not yet. Week ends and working on another projects.

yasirroni, 2022-04-19:
Hi @SanPen, Looking at the [doc][1], it seems that the mentioned DC OPF is for solving lossless assumption of ACOPF since it use voltage angle, whereas DC system has no angle.

[1]: https://gridcal.readthedocs.io/en/latest/theory/opf/opf_dc_ts.html

yasirroni, 2022-08-05:
I've done this a long time ago. Please look at it.

yasirroni, 2022-08-05:
https://github.com/yasirroni/GridCalPlayground

SanPen, 2022-08-05:
What is that I should look at?
It is also not really clear what you want to solve.

yasirroni, 2022-08-05:
I just want to report back that GridCal parsing AC systems from MATPOWER can be editted into DC systems and run power flow on top of it.
_
Basically, I just wanted to reply this:
> Any news?
Yeah, done.
_
And I think, it would be nice if GridCal have a separate repository that anyone can give an working example implementing GridCal. It help development if you got no time to write the example on the docs.

> not really, we're behind with the docs. It should be very intuitive, try it and let me know if you manage.
> 
> br, Santiago

closed 2022-08-05 by yasirroni

SanPen, 2022-08-05:
Alright!
I think it is a good initiative to have a repository just for examples.

I'll post a link to it.