SanPen/VeraGrid #155: Error reading a CIM file

noamarbel, 2022-07-19:
When reading a CIM set of files, I get the following error:
Loading GridCal...
Bentayga is not available
Newton native is not available

Eigen solver is used because MKL runtime is not found.
Alliander's PGM available
Newton Power Analytics is not available: No module named 'newtonpa'
QWindowsWindow::setGeometry: Unable to set geometry 1920x1253+3840+626 (frame: 1936x1292+3832+595) on QWidgetWindow/"mainWindowWindow" on "\\.\DISPLAY3". Resulting geometry: 1920x1009+3840+626 (frame: 1936x1048+3832+595) margins: 8, 31, 8, 8 minimum size: 1351x1253 MINMAXINFO maxSize=0,0 maxpos=0,0 mintrack=1367,1292 maxtrack=0,0)
Traceback (most recent call last):
  File "C:\Users\noama\AppData\Roaming\Python\Python310\site-packages\GridCal\Gui\Session\file_handler.py", line 93, in run
    self.circuit = file_handler.open(text_func=self.progress_text.emit,
  File "C:\Users\noama\AppData\Roaming\Python\Python310\site-packages\GridCal\Engine\IO\file_handler.py", line 72, in open
    self.circuit = parser.load_cim_file(self.file_name)
  File "C:\Users\noama\AppData\Roaming\Python\Python310\site-packages\GridCal\Engine\IO\cim\cim_parser.py", line 994, in load_cim_file
    self.parse_switches(self.cim, circuit, busbar_dict)
  File "C:\Users\noama\AppData\Roaming\Python\Python310\site-packages\GridCal\Engine\IO\cim\cim_parser.py", line 863, in parse_switches
    b1, b2 = elm.get_buses()
AttributeError: 'GeneralContainer' object has no attribute 'get_buses'

The import fails, and non of the files get loaded. 

It is important to note that I am loading a number of files (220 files) that are part of a circuit.

Any ideas on how I can solve this?

Thanks.

SanPen, 2022-07-19:
Hi!

Wow 220 files!

As you may know already CIM is not a standard, but some sort of guidelines, so this may fail for whatever kind of funkyness.

That being said, the error itself is because at that line the object was expected to be a Switch and it is a GeneralContainer instead. Why? no idea. I could check if I had the same input or a meaningful sample.

Would it be possible to share the data to debug?

BR,
Santiago

noamarbel, 2022-07-19:
So I found the one file that causes the issue. I can share it with you for
debugging if you can send me your mail. I cannot post it here on GH.

Thanks.

SanPen, 2022-07-19:
you can send it to me to santiago@gridcal.org

If the file causing the issue is loaded isolated from the others I might not see the same issue as loading all the others...be aware of it.

noamarbel, 2022-07-21:
Thanks for the quick fix and turnaround. I can now read the data!

closed 2022-07-21 by noamarbel