SanPen/VeraGrid #159: XXX is not available

SS4x13, 2022-08-30:
Hi Santiago, 

When I use the command "python -c "from GridCal.ExecuteGridCal import run; run()"" to run GridCal, there are few lines information pop out as below:

Loading GridCal...
Bentayga is not available
Newton native is not available

Eigen solver is used because MKL runtime is not found.
Alliander's PGM available
Newton Power Analytics is not available: No module named 'newtonpa'

---------------------

It is metioning something is not available, does it normal? If it's not, how should I fix it?

Sorry for disturb, I am new to python and may not describe question clearly, hope your patience is not consumed. 
Thank you!

SanPen, 2022-08-30:
Hi!

It is perfectly normal. Those are alternative calculation engines, but none of them provide more functionality. Ussually faster of the same functionality.

Please, always ask any question you may have :D

best regards,
Santiago

SS4x13, 2022-08-31:
Appreciate for your explain!!

closed 2022-08-31 by SS4x13