SanPen/VeraGrid #160: Python >= 3.8

yasirroni, 2022-09-10:
Based on `setup.py`, GridCal require Python <3.8.

Is there any plans to support 3.8 - 3.10, especially Google Colab use Python 3.8.

Furthermore, latest NumPy and Pandas only support >=3.8.

SanPen, 2022-09-10:
Hi!

[This ](https://github.com/SanPen/GridCal/blob/master/src/setup.py) is the actual setup. The one in the main folder is just for CI/CD. So from my experience 3.6 to 3.10 works perfectly. 3.11 we'll see since it'll depends on pyside2.

Best regards,
Santiago

yasirroni, 2022-09-10:
That's good. I can safely migrate to 3.8 then. But in my experience, pytest fail on 3.10 due to depreciation in typing. I'll report back if I found the exact cause. Thank you.

closed 2022-09-10 by yasirroni