SanPen/VeraGrid #171: TypeError: unsupported operand type(s) in GridCal/Engine/Core/Compilers/circuit_to_data2.py:46

cuihantao, 2023-01-28:
Greetings,

With contributions from @JosepFanals, the first step to interopate ANDES with GridCal is finally there! 

I'm looking to troubleshoot a CI error. The issue seems to be related to a very recent version of GridCal. Details can be found in the logs below. Is it a known issue? I can dig up more details in the issue report if needed.

https://github.com/cuihantao/andes/actions/runs/4033471131/jobs/6934010669

JosepFanals, 2023-01-28:
Hi all,

I believe this error could be caused by the union operator `|`, which is defined in Python 3.10, see [PEP 604](https://peps.python.org/pep-0604/). 

Hence the build with Python 3.9 fails.

Could this be the issue, @SanPen? If so, I guess that working with the traditional `Union` would solve it.

SanPen, 2023-01-29:
Hi, 

I've just done a release without the pipe character. Give it a try and let me know

cuihantao, 2023-01-29:
That works. Thank you!

closed 2023-01-29 by cuihantao