SanPen/VeraGrid #178: result different from PowerWorld

weigao-123, 2023-06-20:
case: https://github.com/SanPen/GridCal/blob/master/src/tests/data/grids/IEEE%2030%20bus.raw

lodf result from GridCal (only one column lodf is attached):
<table border="1" class="dataframe">  <thead>    <tr style="text-align: right;">      <th></th>      <th>1_GlenLyn_132.0_2_Claytor_132.0_1</th>    </tr>  </thead>  <tbody>    <tr>      <th>4_Hancock_132.0_6_Roanoke_132.0_1</th>      <td>6.077462e-01</td>    </tr>    <tr>      <th>6_Roanoke_132.0_7_Blaine_132.0_1</th>      <td>1.997210e-01</td>    </tr>    <tr>      <th>4_Hancock_132.0_12_Hancock_33.0_1</th>      <td>4.092236e-02</td>    </tr>    <tr>      <th>16_Bus16_33.0_17_Bus17_33.0_1</th>      <td>1.856839e-02</td>    </tr>    <tr>      <th>12_Hancock_33.0_16_Bus16_33.0_1</th>      <td>1.856839e-02</td>    </tr>    <tr>      <th>12_Hancock_33.0_15_Bus15_33.0_1</th>      <td>1.737964e-02</td>    </tr>    <tr>      <th>15_Bus15_33.0_23_Bus23_33.0_1</th>      <td>1.184819e-02</td>    </tr>    <tr>      <th>23_Bus23_33.0_24_Bus24_33.0_1</th>      <td>1.184819e-02</td>    </tr>    <tr>      <th>19_Bus19_33.0_20_Bus20_33.0_1</th>      <td>1.050578e-02</td>    </tr>    <tr>      <th>18_Bus18_33.0_19_Bus19_33.0_1</th>      <td>1.050578e-02</td>    </tr>    <tr>      <th>15_Bus15_33.0_18_Bus18_33.0_1</th>      <td>1.050578e-02</td>    </tr>    <tr>      <th>24_Bus24_33.0_25_Bus25_33.0_1</th>      <td>7.777503e-03</td>    </tr>    <tr>      <th>25_Bus25_33.0_27_Cloverdl_33.0_1</th>      <td>7.777503e-03</td>    </tr>    <tr>      <th>12_Hancock_33.0_14_Bus14_33.0_1</th>      <td>4.974331e-03</td>    </tr>    <tr>      <th>14_Bus14_33.0_15_Bus15_33.0_1</th>      <td>4.974331e-03</td>    </tr>    <tr>      <th>25_Bus25_33.0_26_Bus26_33.0_1</th>      <td>4.011478e-16</td>    </tr>    <tr>      <th>12_Hancock_33.0_13_Hancock_11.0_1</th>      <td>0.000000e+00</td>    </tr>    <tr>      <th>1_GlenLyn_132.0_3_Kumis_132.0_1</th>      <td>0.000000e+00</td>    </tr>    <tr>      <th>27_Cloverdl_33.0_29_Bus29_33.0_1</th>      <td>0.000000e+00</td>    </tr>    <tr>      <th>3_Kumis_132.0_4_Hancock_132.0_1</th>      <td>0.000000e+00</td>    </tr>    <tr>      <th>27_Cloverdl_33.0_30_Bus30_33.0_1</th>      <td>0.000000e+00</td>    </tr>    <tr>      <th>29_Bus29_33.0_30_Bus30_33.0_1</th>      <td>0.000000e+00</td>    </tr>    <tr>      <th>9_Roanoke_1.0_11_Roanoke_11.0_1</th>      <td>-4.011478e-16</td>    </tr>    <tr>      <th>6_Roanoke_132.0_8_Reusens_132.0_1</th>      <td>-1.543135e-03</td>    </tr>    <tr>      <th>8_Reusens_132.0_28_Cloverdl_132.0_1</th>      <td>-1.543135e-03</td>    </tr>    <tr>      <th>10_Roanoke_33.0_22_Bus22_33.0_1</th>      <td>-1.614181e-03</td>    </tr>    <tr>      <th>10_Roanoke_33.0_21_Bus21_33.0_1</th>      <td>-2.456505e-03</td>    </tr>    <tr>      <th>21_Bus21_33.0_22_Bus22_33.0_1</th>      <td>-2.456505e-03</td>    </tr>    <tr>      <th>22_Bus22_33.0_24_Bus24_33.0_1</th>      <td>-4.070686e-03</td>    </tr>    <tr>      <th>6_Roanoke_132.0_28_Cloverdl_132.0_1</th>      <td>-6.234368e-03</td>    </tr>    <tr>      <th>28_Cloverdl_132.0_27_Cloverdl_33.0_1</th>      <td>-7.777503e-03</td>    </tr>    <tr>      <th>10_Roanoke_33.0_20_Bus20_33.0_1</th>      <td>-1.050578e-02</td>    </tr>    <tr>      <th>6_Roanoke_132.0_10_Roanoke_33.0_1</th>      <td>-1.219026e-02</td>    </tr>    <tr>      <th>10_Roanoke_33.0_17_Bus17_33.0_1</th>      <td>-1.856839e-02</td>    </tr>    <tr>      <th>9_Roanoke_1.0_10_Roanoke_33.0_1</th>      <td>-2.095460e-02</td>    </tr>    <tr>      <th>6_Roanoke_132.0_9_Roanoke_1.0_1</th>      <td>-2.095460e-02</td>    </tr>    <tr>      <th>2_Claytor_132.0_5_Fieldale_132.0_1</th>      <td>-1.997210e-01</td>    </tr>    <tr>      <th>5_Fieldale_132.0_7_Blaine_132.0_1</th>      <td>-1.997210e-01</td>    </tr>    <tr>      <th>2_Claytor_132.0_6_Roanoke_132.0_1</th>      <td>-4.489475e-01</td>    </tr>    <tr>      <th>2_Claytor_132.0_4_Hancock_132.0_1</th>      <td>-6.005189e-01</td>    </tr>    <tr>      <th>1_GlenLyn_132.0_2_Claytor_132.0_1</th>      <td>-1.000000e+00</td>    </tr>  </tbody></table>

lodf result from PowerWorld:
<table border="1" class="dataframe">  <thead>    <tr style="text-align: right;">      <th></th>      <th>BusNum</th>      <th>BusName</th>      <th>BusNum:1</th>      <th>BusName:1</th>      <th>LineCircuit</th>      <th>LineMW</th>      <th>LineLODFMult</th>    </tr>  </thead>  <tbody>    <tr>      <th>0</th>      <td>12</td>      <td>Hancock</td>      <td>13</td>      <td>Hancock</td>      <td>1</td>      <td>0.000</td>      <td>1.000000e+08</td>    </tr>    <tr>      <th>1</th>      <td>9</td>      <td>Roanoke</td>      <td>11</td>      <td>Roanoke</td>      <td>1</td>      <td>0.000</td>      <td>1.000000e+08</td>    </tr>    <tr>      <th>2</th>      <td>25</td>      <td>Bus 25</td>      <td>26</td>      <td>Bus 26</td>      <td>1</td>      <td>3.545</td>      <td>1.000000e+08</td>    </tr>    <tr>      <th>3</th>      <td>3</td>      <td>Kumis</td>      <td>4</td>      <td>Hancock</td>      <td>1</td>      <td>82.210</td>      <td>1.000000e+02</td>    </tr>    <tr>      <th>4</th>      <td>1</td>      <td>Glen Lyn</td>      <td>3</td>      <td>Kumis</td>      <td>1</td>      <td>87.723</td>      <td>1.000000e+02</td>    </tr>    <tr>      <th>5</th>      <td>4</td>      <td>Hancock</td>      <td>6</td>      <td>Roanoke</td>      <td>1</td>      <td>72.147</td>      <td>2.949000e+01</td>    </tr>    <tr>      <th>6</th>      <td>6</td>      <td>Roanoke</td>      <td>7</td>      <td>Blaine</td>      <td>1</td>      <td>38.120</td>      <td>2.208000e+01</td>    </tr>    <tr>      <th>7</th>      <td>4</td>      <td>Hancock</td>      <td>12</td>      <td>Hancock</td>      <td>1</td>      <td>44.205</td>      <td>6.190000e+00</td>    </tr>    <tr>      <th>8</th>      <td>12</td>      <td>Hancock</td>      <td>16</td>      <td>Bus 16</td>      <td>1</td>      <td>7.249</td>      <td>2.950000e+00</td>    </tr>    <tr>      <th>9</th>      <td>16</td>      <td>Bus 16</td>      <td>17</td>      <td>Bus 17</td>      <td>1</td>      <td>3.696</td>      <td>2.950000e+00</td>    </tr>    <tr>      <th>10</th>      <td>15</td>      <td>Bus 15</td>      <td>23</td>      <td>Bus 23</td>      <td>1</td>      <td>5.039</td>      <td>2.310000e+00</td>    </tr>    <tr>      <th>11</th>      <td>23</td>      <td>Bus 23</td>      <td>24</td>      <td>Bus 24</td>      <td>1</td>      <td>1.808</td>      <td>2.310000e+00</td>    </tr>    <tr>      <th>12</th>      <td>25</td>      <td>Bus 25</td>      <td>27</td>      <td>Cloverdl</td>      <td>1</td>      <td>-4.758</td>      <td>2.240000e+00</td>    </tr>    <tr>      <th>13</th>      <td>24</td>      <td>Bus 24</td>      <td>25</td>      <td>Bus 25</td>      <td>1</td>      <td>-1.204</td>      <td>2.240000e+00</td>    </tr>    <tr>      <th>14</th>      <td>18</td>      <td>Bus 18</td>      <td>19</td>      <td>Bus 19</td>      <td>1</td>      <td>2.780</td>      <td>2.030000e+00</td>    </tr>    <tr>      <th>15</th>      <td>19</td>      <td>Bus 19</td>      <td>20</td>      <td>Bus 20</td>      <td>1</td>      <td>-6.727</td>      <td>2.030000e+00</td>    </tr>    <tr>      <th>16</th>      <td>15</td>      <td>Bus 15</td>      <td>18</td>      <td>Bus 18</td>      <td>1</td>      <td>6.020</td>      <td>2.030000e+00</td>    </tr>    <tr>      <th>17</th>      <td>12</td>      <td>Hancock</td>      <td>15</td>      <td>Bus 15</td>      <td>1</td>      <td>17.899</td>      <td>1.480000e+00</td>    </tr>    <tr>      <th>18</th>      <td>14</td>      <td>Bus 14</td>      <td>15</td>      <td>Bus 15</td>      <td>1</td>      <td>1.586</td>      <td>6.300000e-01</td>    </tr>    <tr>      <th>19</th>      <td>12</td>      <td>Hancock</td>      <td>14</td>      <td>Bus 14</td>      <td>1</td>      <td>7.859</td>      <td>6.300000e-01</td>    </tr>    <tr>      <th>20</th>      <td>27</td>      <td>Cloverdl</td>      <td>29</td>      <td>Bus 29</td>      <td>1</td>      <td>6.190</td>      <td>0.000000e+00</td>    </tr>    <tr>      <th>21</th>      <td>27</td>      <td>Cloverdl</td>      <td>30</td>      <td>Bus 30</td>      <td>1</td>      <td>7.092</td>      <td>0.000000e+00</td>    </tr>    <tr>      <th>22</th>      <td>29</td>      <td>Bus 29</td>      <td>30</td>      <td>Bus 30</td>      <td>1</td>      <td>3.703</td>      <td>0.000000e+00</td>    </tr>    <tr>      <th>23</th>      <td>10</td>      <td>Roanoke</td>      <td>22</td>      <td>Bus 22</td>      <td>1</td>      <td> [... cut ...]     <td>Claytor</td>      <td>4</td>      <td>Hancock</td>      <td>1</td>      <td>43.622</td>      <td>-3.817000e+01</td>    </tr>    <tr>      <th>40</th>      <td>1</td>      <td>Glen Lyn</td>      <td>2</td>      <td>Claytor</td>      <td>1</td>      <td>173.228</td>      <td>-1.000000e+02</td>    </tr>  </tbody></table>

Both results look resonable, but the values are different, could you please help to investigate it? Please let me know if there is any other info I can provide. Thank you!

weigao-123, 2023-06-20:
@weigao-123 By the way, result from PW is in the unit of percentage and any value grater than 100 should be just 0

SanPen, 2023-06-20:
Hi!

My first reaction, would be: Is power world LODF using the "AC" LODF derived using the jacobian, or is the one using the "DC" linear matrices?

A side note is that the branch ordering might not be the same...

We've tested the LODF against PSSe but not Power world. In general LODF's produce mathematical glitches that Power World might correct differently from us.

Regardless, the best possible test is to compute the LODF with power flows and see who is "more correct".

Let me know if you need any assistance.

weigao-123, 2023-06-20:
Thank you!

I also used "Losses DC" method for the LODF calculation.
Actually, the branch ordering alignment is done on my own, and so if we look a specific line

**GridCal: lodf(1_GlenLyn_132.0_2_Claytor_132.0_1, 4_Hancock_132.0_6_Roanoke_132.0_1) = 0.6077462**
**PowerWorld: lodf(..., ...) = 0.2949000**

Power flow testing for line flow "1_GlenLyn_132.0_2_Claytor_132.0_1" at outage "4_Hancock_132.0_6_Roanoke_132.0_1":
- GridCal (Solver: Linear DC, frequency=60 Hz):
    - pre_outage: 1_GlenLyn_132.0_2_Claytor_132.0_1 = 161.026347 MW, 4_Hancock_132.0_6_Roanoke_132.0_1 = 72.424085 MW
    - post_outage: 1_GlenLyn_132.0_2_Claytor_132.0_1 = 182.200912 MW
    - **lodf: (182.200912-161.026347)/72.424085 = 0.29237**
    
- PowerWorld (Solver: DC Power Flow, frequency=60 Hz):
    - pre_outage: 1_GlenLyn_132.0_2_Claytor_132.0_1 = 161.08 MW, 4_Hancock_132.0_6_Roanoke_132.0_1 =  73.11 MW
    - post_outage: 1_GlenLyn_132.0_2_Claytor_132.0_1 = 182.64 MW
    - **lodf: (182.64-161.08)/73.11 = 0.29490**
    
From this, I noticed maybe the row and column of LODF matrix in GridCal is transposed compared with PowerWorld; in PowerWorld, each row reprensents the outaged line, and each column reprensts the monitored line, but in GridCal, each row reprensents the monitored line, and each column reprensts the outaged line, am I correct?

SanPen, 2023-06-20:
Yes, you are correct. The rows are the monitored branches and the columns is waht changes, for the PTDF the changes are the bus injections and for the LODF the changes are the branch outages.
Both have consistent dimentions on that sense.

weigao-123, 2023-06-20:
@SanPen Make sense, thank you so much!

closed 2023-06-22 by SanPen