SanPen/VeraGrid #179: Inductive constant impedance loads from .raw files

JosepFanals, 2023-07-03:
Hi Santiago,

People from CITCEA have been relying on GridCal to perform some validations of the PSSE power flows. There were some minor discrepancies in the results, which is believed to be due to the constant impedance load model. As far as I know, PSSE provides you with the power `(YPload, YQload)` derived from the constant impedance under nominal voltage conditions. Departing from $\underline{S} = V^2 \underline{Y}^*$, if we further develop it, we get $P+jQ = V^2 (G-jB)$. This implies the sign of $B$ has to be the contrary of that of $Q$. So following this trace of thought, if we go to line 409 of the `raw_parser.py` file, we should modify it to `b = - self.YQ`. 

I am now thinking that this may have been the cause behind the slightly different results we obtained when simulating the ACTIVSg2000 system some time ago. 

Anyway, if you can double check it to be the case, I would greatly appreciate it. 

All credit goes to @eliamateu for spotting this potential bug.

SanPen, 2023-07-03:
alright! I'll fix this for the next 5.0 release

JosepFanals, 2023-07-03:
I am not sure this is the best approach. I will try to explain it as best as I can.

First, in https://github.com/SanPen/GridCal/commit/f8219368ee0c06e26f7159288aef07fd824d15f2 it seems to me you have coded it as `S0 + I0 * Vm + np.conj(Y0) * np.power(Vm, 2)` but by this logic to be coherent you may also have to conjugate the current, i.e. ``S0 + np.conj(I0) * Vm + np.conj(Y0) * np.power(Vm, 2)`` (which is what you have written in the comment above). 

In either case, current sources and constant impedances are given in units of power in PSSE. Let's reason it by defining the nomenclature: 

- Power from current sources: $\underline{S}_i = P_i + jQ_i = \underline{V} \underline{I}^* =  \underline{V} (I_r - jI_i)$.
- Power from shunt impedances: $\underline{S}_s = P_s + jQ_s = V^2 \underline{Y}^* = V^2 (G - jB)$. 

Now, the values parsed from PSSE are $P_i$, $Q_i$, $P_s$ and $Q_s$. This implies the signs of the imaginary parts should be modified in principle to make physical sense. For example, if you want `b` to represent a susceptance, then `b = -self.YQ`. I can understand you could wish to keep the inputs as they are, which is feasible to do, but in that case there should be no complex conjugate in `compute_zip_power()`! This is because the power should be directly ``S0 + I0 * Vm + Y0 * np.power(Vm, 2)`` as `I0` and `Y0` are the actual powers from PSSE. I think this is what you previously had coded [here](https://github.com/SanPen/GridCal/blob/f702824ce282e1d1d1366aec041be5b002c0a31d/src/GridCal/Engine/Simulations/PowerFlow/NumericalMethods/common_functions.py#L98), am I right?

I do not know the implications that modifying the `compute_zip_powers()` could have, especially if it is being used for models where the shunt impedances and current sources of the load model are not defined as in PSSE.

SanPen, 2023-07-03:
I think that the formula should reflect the canonical operation: `S0 + np.conj(I0) * Vm + np.conj(Y0) * np.power(Vm, 2)` that simplifies to `S0 + np.conj(Y0 * Vm + I0) * Vm`. Then the PSSe parser should remain as is, meaning no negative sign.

For the general case, aside from the RAW parser, the previous formula `S0 + I0 * Vm + Y0 * np.power(Vm, 2)` is incorrect, and should contain the conjugates, agree?

JosepFanals, 2023-07-03:
Yep, I agree on the fact that the way to go would be to employ `S0 + np.conj(I0) * Vm + np.conj(Y0) * np.power(Vm, 2)`. In this formula, `I0` and `Y0` should have units of currents and admittances respectively imo. If not, you are inputting powers and then changing the sign of their imaginary part. Hopefully you can take a look at the manual so we can all have a clear understanding of what is going on. I also think it should not be that hard!

SanPen, 2023-07-03:
So, I've done the following experiment:

I have this very simple model: [ZIP load example.raw.txt](https://github.com/SanPen/GridCal/files/11937077/ZIP.load.example.raw.txt)

PSSe:

|       | Vm (p.u.) | Va (deg) |
|-------|-----------|----------|
| Bus1  | 1.0       | 0.0      |
| Bus P | 0.9893    | -5.1287  |
| Bus I | 0.9856    | -9.1535  |
| bus Y | 0.9858    | -11.4464 |

In PSSe what appear on the loads table is:

|        | P   | Q | IP  | IQ | YP  | YQ |
|--------|-----|---|-----|----|-----|----|
| Load P | 200 | 5 | 0   | 0  | 0   | 0  |
| Load I | 0   | 0 | 300 | 7  | 0   | 0  |
| Load Y | 0   | 0 | 0   | 0  | 400 | 10 |

In GridCal:

Leaving the import values the same, and modifying the formula to `S0 + np.conj(Y0 * Vm + I0) * Vm` the voltages are:

|       | Vm (p.u.) | Va (deg) |
|-------|-----------|----------|
| Bus1  | 1.0       | 0.0      |
| Bus P | 0.9907   | -5.1361 |
| Bus I | 0.9883    | -9.1587  |
| bus Y | 0.9885    | -11.4493 |

I would say, it's close enough.

SanPen, 2023-07-04:
Do we have any larger grid to test this with?

JosepFanals, 2023-07-04:
Hi Santiago,

From what I know, the issue was identified in a variation of the IEEE 118-bus system. Changing the signs in the parser caused the power flow results to coincide. Of course the right solution is to add the conjugates in the function, which would then yield the same results. 

Strangely enough the results in this simple grid are quite different, and we are checking it first. What tolerances did you employ in PSSE and GridCal? We could also increase the $X$ of the lines by a factor of 5 maybe to notice larger differences.

JosepFanals, 2023-07-05:
I am quite confident we have found where the source of the issue. Let me explain:

- The function `compute_zip_powers()` has to have the shunt current and admittance conjugated as we have been discussing, there is no doubt in that. 
- The only thing to modify in the `raw_parser.py` is the sign of the imaginary part of the current (`self.IQ` has to be replaced by `-self.IQ`). While the PSSE manual indicates that the sign of `YQ` is automatically modified, there is no comment on having changed the sign of `IQ`. 

[image]
[image]

This explains why in your very simple 4-bus system we were not able to get the right solution, whereas yesterday we got matching results for the modified IEEE 118-bus system (only having introduced the complex conjugates in the power calculation). 

Considering the two modifications that I have listed above, I get the exact same results as PSSE in the 4-bus system: 

<html>
<body>

Bus  |  Vm (p.u.) | Va (deg)
-- | -- | --
Bus1 | 1.0 | 0.0
Bus P | 0.98932775 | -5.12870578
Bus I | 0.98559523 | -9.15352654
bus Y | 0.98579192 | -11.44642704

</body>
</html>

What do you think? I feel like the issue is solved with these two changes. We can privately send you the modified IEEE 118-bus system if you want to play with them.

SanPen, 2023-07-05:
I have implemented the changes in df503ec11a061c7d0860e7c4700ea55dea43d2a4

If you can share the IEEE118 with the changes publicly, I'd like to create a test for it.

JosepFanals, 2023-07-06:
Hi Santiago,

I attach two files of the modified IEEE118 system. Loads are treated as PQ in one of them, and as ZI in the other one. 

[IEEE118busWinter Solved_mod_PQ_gc.raw.txt](https://github.com/SanPen/GridCal/files/11965624/IEEE118busWinter.Solved_mod_PQ_gc.raw.txt)

[IEEE118busWinter Solved_mod_IZ_gc.raw.txt](https://github.com/SanPen/GridCal/files/11965627/IEEE118busWinter.Solved_mod_IZ_gc.raw.txt)

Feel free to create a couple of tests with them.

closed 2023-09-29 by SanPen