SanPen/VeraGrid #237: Error when running on MacOS with Apple Silicon cpu

PIC-Lab, 2023-11-23:
python3 -c "from GridCal.ExecuteGridCal import run; run()"
/Users/<username>/Library/Python/3.9/lib/python/site-packages/networkx/utils/backends.py:135: RuntimeWarning: networkx backend defined more than once: nx-loopback
  backends.update(_get_backends("networkx.backends"))
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/<username>/Library/Python/3.9/lib/python/site-packages/GridCal/ExecuteGridCal.py", line 21, in <module>
    from GridCal.Gui.Main.GridCalMain import run
  File "/Users/<username>/Library/Python/3.9/lib/python/site-packages/GridCal/Gui/Main/GridCalMain.py", line 20, in <module>
    from GridCal.Gui.GridEditorWidget import *
  File "/Users/<username>/Library/Python/3.9/lib/python/site-packages/GridCal/Gui/GridEditorWidget.py", line 21, in <module>
    from PIL.ImageQt import ImageQt
ImportError: cannot import name 'ImageQt' from 'PIL.ImageQt' (/Users/<username>/Library/Python/3.9/lib/python/site-packages/PIL/ImageQt.py)

SanPen, 2023-11-23:
Hi!

I have just made some adjustments to the setup.

Please upgrade:

```
pip install GridCal --upgrade
```

Then you should be able to run it by typing `gridcal` on the terminal.

I'm having a bit of hell myself on my mac M1 because I have anaconda as my main python (which is a terrible idea altogether) Regularly I develop using python 3.10 or 3.11 installed with brew.

Let me know if this solves it for you.

PIC-Lab, 2023-11-27:
Great, now I can run it. Thank you！

closed 2023-11-27 by SanPen