SanPen/VeraGrid #261: Potential issue in get_bus_branch_connectivity_matrix()

JosepFanals, 2024-02-06:
Hi Santiago,

Francesca Rossi, a PhD student at CITCEA, noticed the `get_bus_branch_connectivity_matrix()` method may not work as expected. The thing is, `branch_list` groups branch data for lines, transformers, etc. Therefore, the row index of `Cf` and `Ct` may be erroneous since the `k` index is not cumulative. I am pasting below a fraction of the original code (L3490 of `multi_circuit.py`):

```
        for branch_list in self.get_branch_lists():
            for k, br in enumerate(branch_list):
                i = bus_dict[br.bus_from]  # store the row indices
                j = bus_dict[br.bus_to]  # store the row indices
                Cf[k, i] = 1
                Ct[k, j] = 1
```

Instead, Francesca proposes to modify it for: 

```
        k_len_cum = 0
        for branch_list in self.get_branch_lists():
            for k_1, br in enumerate(branch_list):
                k = k_1 + k_len_cum
                i = bus_dict[br.bus_from]  # store the row indices
                j = bus_dict[br.bus_to]  # store the row indices
                Cf[k, i] = 1
                Ct[k, j] = 1
            k_len_cum += len(branch_list)
```

From what I could check, the test suite runs equally on both cases, but I do believe the second block of code is the right one. What do you think?

SanPen, 2024-02-07:
Hi, indeed there was a bug and I've just fixed it in `devel` with [e197cd8](https://github.com/SanPen/GridCal/commit/e197cd8c868113954cee9b62f4a25fc019f4adfc).

However this function is not used for anything critical because it "lives" in the `MultiCircuit`.

Please advise Francesca that she should use `Cf` and `Ct` from the `NumericalCircuit` instead. Those account for branch and bus active and are the ones used for numerical processes.

closed 2024-02-07 by SanPen

francesca16, 2024-02-07:
Hi Josep, Hi Santiago,

Thank you for fixing the bug and for the clarification!