SanPen/VeraGrid #271: Bug when running a Powerflow

fernpos, 2024-04-01:
Traceback (most recent call last):
  File "C:\ReposGit\github\fernpos\GridCal5\src\GridCal\Session\session.py", line 221, in run
    self.driver.run()
  File "C:\ReposGit\github\fernpos\GridCal5\src\GridCalEngine\Simulations\PowerFlow\power_flow_ts_driver.py", line 207, in run
    self.results = self.run_single_thread(time_indices=self.time_indices)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\ReposGit\github\fernpos\GridCal5\src\GridCalEngine\Simulations\PowerFlow\power_flow_ts_driver.py", line 96, in run_single_thread
    pf_res = pf_worker.multi_island_pf(multi_circuit=self.grid,
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\ReposGit\github\fernpos\GridCal5\src\GridCalEngine\Simulations\PowerFlow\power_flow_worker.py", line 738, in multi_island_pf
    nc = compile_numerical_circuit_at(
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\ReposGit\github\fernpos\GridCal5\src\GridCalEngine\DataStructures\numerical_circuit.py", line 1942, in compile_numerical_circuit_at
    nc.generator_data, gen_dict = gc_compiler2.get_generator_data(circuit=circuit,
                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\ReposGit\github\fernpos\GridCal5\src\GridCalEngine\Compilers\circuit_to_data.py", line 333, in get_generator_data
    i = bus_dict[elm.bus]
        ~~~~~~~~^^^^^^^^^
KeyError: 'None'

SanPen, 2024-04-01:
En qué red?  En general hay que detallar los pasos para reproducir el bug

SanPen, 2024-04-03:
El bug venía que los bus_prof se quedaban a None. Ahora al establecer el bus del snapshot, se rellena el perfil.