SanPen/VeraGrid #272: Installing and running on mac M1 - macOS Ventura

mrsilvabr, 2024-04-02:
Hi
I have got the following error lines, after running pip install GridCal or pip install GridCal --upgrade

maurosilva@Mauros-MacBook-Air ~ % gridcal
Traceback (most recent call last):
  File "/Users/maurosilva/anaconda3/bin/gridcal", line 5, in <module>
    from GridCal.ExecuteGridCal import runGridCal
  File "/Users/maurosilva/anaconda3/lib/python3.11/site-packages/GridCal/ExecuteGridCal.py", line 23, in <module>
    from GridCal.Gui.Main.GridCalMain import runGridCal
  File "/Users/maurosilva/anaconda3/lib/python3.11/site-packages/GridCal/Gui/Main/GridCalMain.py", line 24, in <module>
    from GridCal.Gui.Main.SubClasses.Scripting.scripting import ScriptingMain
  File "/Users/maurosilva/anaconda3/lib/python3.11/site-packages/GridCal/Gui/Main/SubClasses/Scripting/scripting.py", line 20, in <module>
    from GridCal.Gui.Main.SubClasses.io import IoMain
  File "/Users/maurosilva/anaconda3/lib/python3.11/site-packages/GridCal/Gui/Main/SubClasses/io.py", line 33, in <module>
    from GridCal.Gui.RosetaExplorer.RosetaExplorer import RosetaExplorerGUI
  File "/Users/maurosilva/anaconda3/lib/python3.11/site-packages/GridCal/Gui/RosetaExplorer/RosetaExplorer.py", line 23, in <module>
    from GridCal.Gui.RosetaExplorer.ConsoleWidget import ConsoleWidget
  File "/Users/maurosilva/anaconda3/lib/python3.11/site-packages/GridCal/Gui/RosetaExplorer/ConsoleWidget.py", line 2, in <module>
    from qtconsole.rich_jupyter_widget import RichJupyterWidget
  File "/Users/maurosilva/anaconda3/lib/python3.11/site-packages/qtconsole/rich_jupyter_widget.py", line 9, in <module>
    from qtpy import QtCore, QtGui, QtWidgets
  File "/Users/maurosilva/anaconda3/lib/python3.11/site-packages/qtpy/QtCore.py", line 112, in <module>
    Qt.MidButton = Qt.MouseButton.MiddleButton = Qt.MiddleButton
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/maurosilva/anaconda3/lib/python3.11/enum.py", line 836, in __setattr__
    raise AttributeError('cannot reassign member %r' % (name, ))
AttributeError: cannot reassign member 'MiddleButton'

Thank you for any help

SanPen, 2024-04-05:
@mrsilvabr any feedback?

mrsilvabr, 2024-04-06:
Hi SanPen.
I tried with pip3 as you said but with the same result. I was unable to try outside Anaconda.
I am unable to di this "using the system python". I simply don't know how to do it. Should I have a "system python"? 
Thank you anyway

mrsilvabr, 2024-04-08:
Hi
I answered in github
Hi SanPen.
I tried with pip3 as you said but with the same result. I was unable to try outside Anaconda.
I am unable to di this "using the system python". I simply don't know how to do it. Should I have a "system python"?
Thank you anyway

SanPen, 2024-04-09:
Hi, 

Yes, MacOS comes with python out of the box. The issue is that by installing Anaconda you've shadowed the python from the OS with Anaconda :/

Apparently you can create an environment in anaconda and install PySide6 and not qtpy. 
My honest advice is to get rid of anaconda. Uninstall it and go back to the system python.

mrsilvabr, 2024-04-09:
Thanks SanPen
I will try it later this week. But what I understood from a variety of sources is that macOS -M1 (my machine) did not come with python installed. Somehow Apple ditch Python from its computers a couple of years ago.

SanPen, 2024-04-09:
That could be the case...Anyway, I'd highly recommend to install homebrew and then python with homebrew.
I'd advise to try the enviroments solution first.

mrsilvabr, 2024-04-10:
Hi SanPen

I unistalled Anaconda and started anew with Homebrew and a virtual environment.

I got this:

"Successfully built gridcal GridCalEngine pandas
Installing collected packages: xlwt, pytz, py4j, xlrd, wheel, tzdata, tqdm, threadpoolctl, shiboken6, setuptools, pyqtdarktheme, pyproj, pyarrow, protobuf, pluggy, nptyping, networkx, joblib, isodate, iniconfig, immutabledict, h5py, geographiclib, future, et-xmlfile, darkdetect, cloudpickle, chardet, absl-py, scikit-learn, rdflib, pytest, PySide6-Essentials, pandas, openpyxl, hyperopt, geopy, windpowerlib, PySide6-Addons, pvlib, ortools, PySide6, GridCalEngine, gridcal
Successfully installed GridCalEngine-5.1.4 PySide6-6.7.0 PySide6-Addons-6.7.0 PySide6-Essentials-6.7.0 absl-py-2.1.0 chardet-5.2.0 cloudpickle-3.0.0 darkdetect-0.8.0 et-xmlfile-1.1.0 future-1.0.0 geographiclib-2.0 geopy-2.4.1 gridcal-5.1.4 h5py-3.11.0 hyperopt-0.2.7 immutabledict-4.2.0 iniconfig-2.0.0 isodate-0.6.1 joblib-1.4.0 networkx-3.3 nptyping-2.5.0 openpyxl-3.1.2 ortools-9.9.3963 pandas-2.2.2 pluggy-1.4.0 protobuf-5.26.1 pvlib-0.10.4 py4j-0.10.9.7 pyarrow-15.0.2 pyproj-3.6.1 pyqtdarktheme-0.1.7 pytest-8.1.1 pytz-2024.1 rdflib-7.0.0 scikit-learn-1.4.2 setuptools-69.2.0 shiboken6-6.7.0 threadpoolctl-3.4.0 tqdm-4.66.2 tzdata-2024.1 wheel-0.43.0 windpowerlib-0.2.2 xlrd-2.0.1 xlwt-1.3.0
(venv) maurosilva@Mauros-MacBook-Air ~ % gridcal
Traceback (most recent call last):
  File "/Users/maurosilva/path/to/venv/bin/gridcal", line 8, in <module>
    sys.exit(runGridCal())
             ^^^^^^^^^^^^
  File "/Users/maurosilva/path/to/venv/lib/python3.12/site-packages/GridCal/Gui/Main/GridCalMain.py", line 124, in runGridCal
    qdarktheme.enable_hi_dpi()
    ^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'qdarktheme' has no attribute 'enable_hi_dpi'
(venv) maurosilva@Mauros-MacBook-Air ~ % "

I tried closing terminal and reopening it, with the same result.

Regards

SanPen, 2024-04-11:
`pyqtdarktheme` is in version 2.1.0, but for some reason pip installed version 0.1.7... Try updating that one

mrsilvabr, 2024-04-12:
Hi SanPen
I downgraded my python installation to 3.9.3 and managed to install GridCal within a virtual environment.  
I consider the issue closed.
Thank you

closed 2024-04-12 by mrsilvabr

SanPen, 2024-04-13:
If you feel like, you could update securely up to 3.11. 
Python 3.11 has a large performance improvement .

If you need further assistance, or find bugs, please let me know.

mrsilvabr, 2024-04-16:
Hi SanPen

I am trying but unsuccessfully.

I inserted one busbar and 1 transformer in the screen.

I clicked in the busbar. The pointer has turned into a finger and stays like that no matter which keyboard action I take.

Regards

￼

SanPen, 2024-04-16:
you should not use anything other than bus, for the time being.

Please, for different issues, create a new issue since this one was closed.