SanPen/VeraGrid #300: License change

SanPen, 2024-08-28:
# Proposal

This issue proposes the re-licensing of GridCal to a more permissive license, while improving the legal coverage.

## Current license: GNU Lesser General Public License (LGPL) 

- **Type**: Weak Copyleft
- **Key Features**: Allows the software to be used in proprietary applications as long as modifications to the LGPL-covered code itself are released under LGPL. Focuses on libraries or components.

### Benefits:

- Copyleft with Flexibility: Ensures that modifications to the LGPL-covered parts of GridCal are open-source, while allowing it to be used in proprietary software. Encourages contributions back to the LGPL-licensed components.

- Integration: Suitable if GridCal is intended to be used as a library or component within larger software systems, including proprietary ones.

### Drawbacks:

- Complex Compliance: Requires that any modifications to the LGPL-covered code be open-source, which can add complexity to project management.

- No Patent Protection: Lacks explicit patent protection, which might expose users to patent claims related to the LGPL code.

### Best For:

Projects where you want to ensure that any modifications to GridCal itself are open-source while allowing it to be used in proprietary applications. Ideal if GridCal is used as a library or component rather than as a standalone application.

## Proposed license: Apache License 2.0

- **Type**: Permissive
- **Key Features**: Allows for broad use, modification, and distribution with minimal restrictions. Includes an explicit patent grant.

### Benefits:

- Permissiveness: Enables easy integration with proprietary software. Ideal if you want GridCal to be used in a wide range of environments, including commercial applications.
  Patent Protection: Provides protection against patent claims from contributors, reducing legal risk.
  Flexibility: Permits inclusion in proprietary projects without requiring the entire project to be open-source.
  Clarity: Clear and straightforward license terms make compliance easier.

### Drawbacks:

- No Copyleft: Does not require modifications to GridCal to be shared, which might result in fewer contributions back to the project.

- Complex Projects: While the license itself is simple, integrating with proprietary systems might involve additional legal considerations.

### Best For

Projects where you want broad adoption, including commercial use, with minimal restrictions on how the software is used and integrated.

## What other projects use:

| Software                                        | License                          |
|-------------------------------------------------|----------------------------------|
| PSAT (Power System Analysis Toolbox)            | GNU General Public License (GPL) |
| GridLAB-D                                       | Custom permissive license        |
| OpenIPSL (Open Integrated Power System Library) | BSD-3-Clause license             |
| MATPOWER                                        | BSD-3-Clause license             |
| PyPSA (Python for Power System Analysis)        | MIT license                      |
| pandapower                                      | Custom permissive license        |
| Powsybl                                         | MPL-2.0 license                  |
| Power-grid-model                                | MPL-2.0 license                  |

SanPen, 2024-08-28:
My personal comment; 

In choosing MPL, more people from industry could be brought into the project since that provided a more comfortable setting than LGPL, while having a similar protection. Derivative works must retain the MPL license too and there is no restriction of the way it is included in other works since the MPL is a weak Copyleft license.

Other licenses that are not copyleft such as MIT, Apache or BSD leave the possibility that some parties make derivatives of GridCal and make it closed source which in my view would be a terrible thing.

poypoyan, 2024-08-29:
No problem for me. Thanks!

JosepFanals, 2024-08-29:
Fine for me as well

miek770, 2024-08-29:
Likewise, no problem for me.

QuimMoya, 2024-08-29:
No problem !!!

Carlos-Alegre, 2024-08-29:
No problem for me!

SanPen, 2024-08-29:
I agree to relicense to MPL 2.0

yasirroni, 2024-08-29:
I agree with the changes. It make GridCal allowed to be statically included on other project.

But, it is worth noting that there will be an edge case explained in SO that user will be able to add functionality while not share the functionality itself.

> someone could take one of the main file of your project, add "import my_private_new_file", and modify your main method for example by adding "my_private_new_file.newAwesomeFeature.run()".

See here: https://softwareengineering.stackexchange.com/a/400544/405045

BegoSanjuan, 2024-08-29:
No problem for me. Thank you!

ClaudiaMachadoCervera, 2024-08-29:
Hi Santiago,

No problem from my side.

Regards,

Claudia Machado Cervera

*Power Systems Engineer*

eRoots Analytics

Carrer del Torrent d'en Vidalet 55 L1

Barcelona, 08024

***@***.*** ***@***.***>

www.eroots.tech
  [image: Photo]

<https://www.linkedin.com/company/eroots-analytics/>
<https://eroots.tech/>

jozsefgorcs-navitasoft, 2024-08-29:
It's okay for me!

eyllanesc, 2024-08-30:
It's okay for me!

peterkulik-navitasoft, 2024-08-30:
Certainly, I agree.

RanjeetHambire, 2024-08-30:
Its OK For me Also!

Bengt, 2024-08-30:
Yes, relicensing to a more permissive license is fine by me. Thanks for the initiative.

jag0nzalez, 2024-09-02:
I agree!

cristinafray, 2024-09-02:
I agree!

jsolerIREC, 2024-09-06:
I agree to the license change!

JanaSoler

jahanbani, 2024-09-11:
Agreed.

leeraiyan, 2024-09-12:
Agreed!

ManuelNvro, 2024-09-30:
I agree!

fernpos, 2024-10-31:
I agree!

ramferan, 2024-10-31:
I agree!

adrisanchu, 2024-10-31:
Agreed!

cggaray, 2024-10-31:
Agreed!

SanPen, 2024-10-31:
Dear all, thank you for taking the time to review and accept this proposal, and thank you very much for your contributions.
The next GridCal release will be done under the MPL license.

closed 2024-10-31 by SanPen