SanPen/VeraGrid #306: Adding wire to Tower

yns0671, 2024-09-12:
I am trying to impelent your distribution grid tutorial but i cannot ad wire to tower. I do same steps as you. When i click the edit tower; wire resistance value is blank, and even if I correct it, i cant add wire to tower.
**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Database'
2. Click on 'Wire', add wire, change parameters of the wire, 
3. Click 'Tower', add tower, right click>edit, cannot see the wire parameters which i changed and cannot add wire to tower

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows10
 - Version 22H

**Additional context**
Add any other context about the problem here.

SanPen, 2024-09-12:
Should be fixed in version 5.1.27!  
Thank you for reporting.

Please let me know if it works for you.

@ClaudiaMachadoCervera @JosepFanals

yns0671, 2024-09-12:
It is working now! Thank you very much.

commit b8810654 (2024-09-12): Fixed #306: wires and tower calculation broke
closed 2024-09-12 by yns0671