SanPen/VeraGrid #307: New vs Old Interface

yns0671, 2024-09-13:
**Describe the bug**
I cant find the ‘Assign selected values to the selected time slot to the grid’ button. Sice the tutorials are old, some features are changed. I could not find how to do it. (Step 10 at the Distribution Grid tutorial)
Steps to reproduce the behavior:
1. Go to 'Database > Load'

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 22H

yns0671, 2024-09-13:
Also, how can we run time series powerr flow.

SanPen, 2024-09-13:
Hi, There are some UI changes indeed...

To run any simulation in time series mode, you need to active the time series mode with this button:
[image]

The button to assign the time series step to the snapshot is located in Database:
[image]

I hope it helps!

closed 2024-09-13 by yns0671