SanPen/VeraGrid #310: Can not load template values for lines

yns0671, 2024-09-16:
**Describe the bug**
When I click on tower and change its name on edit menu, after i click accept and close edit menu, i see that its name did not change. Moreover, when i try to load this tower as template value for the lines, i cant see the created tower. I cannot load template for lines.

**Screenshots**
[image]
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 22H

yns0671, 2024-09-16:
Also, can we change scale on diagram from megawatt to kilowatt for example?

SanPen, 2024-09-16:
> Also, can we change scale on diagram from megawatt to kilowatt for example?

In general not. This may be done in the future but it is quite a lot of work to change everything to display only the power values / 1000...

yns0671, 2024-09-16:
I can see the tower on the associations of the line, but I cannot add the parameters of the tower to the lines on the diagram.
Also, i cannnot load template values when i double click on the lines at the diagram.
[image]

SanPen, 2024-09-16:
What is the `Vnom` of the tower?
What is the `Vnom` of the buses where the line is connected?
Those values should be within 10% difference, and ideally the same.

yns0671, 2024-09-16:
thanks for the clarification. I can do this on the diagram one by one . How can i load the template values for all lines at once on the database?

SanPen, 2024-09-16:
Do you mean, set a templates to all lines?
Your best chance is using a little scripting:

``` python
for line in app.circuit.lines:
    line.apply_template(obj=my_template, Sbase=100.0)
```

yns0671, 2024-09-16:
thank you very much

closed 2024-09-16 by yns0671