SanPen/VeraGrid #311: Per  Unit usceptance

yns0671, 2024-09-19:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Windows 10, Ubuntu 22.04, ...]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

emadellaloggia, 2024-09-19:
maybe this will help
https://mega.co.nz/#!9vE0jIrD!nkSuAAbrT1Sjn01VjaQfVQGAFLqvdIF4EkMeSLuKCWw
I put the necessary dlls in the archive

SanPen, 2024-09-19:
??

SanPen, 2024-09-19:
If you double click on a line you can enter values in per-unit.
It is necessary that you have the line buses voltage properly set

yns0671, 2024-09-19:
When i enter per unit values, this time susceptance value is changed for example from 395 to 0.000395. it may be a problem with units.

SanPen, 2024-09-19:
what are the bus voltages?

yns0671, 2024-09-19:
230 kV

SanPen, 2024-09-20:
Fixed it and unified with a function of the lines called `fill_design_parameters`.
Should be fine in 5.1.37

closed 2024-09-20 by yns0671