SanPen/VeraGrid #313: Digsilent to Gridcal

yns0671, 2024-09-26:
Hello,

I am currently working with DIgSILENT PowerFactory and would like to transition to GridCal for power system analysis. However, I have several models (including diagrams and grid data) in DIgSILENT format. Is there a way to import or convert these files into GridCal, or any recommended workflow for handling such transitions?

Any guidance or tools to facilitate this process would be greatly appreciated.

Thank you in advance!

Best regards,

SanPen, 2024-09-26:
Hi, 

To pass the models, depending on the level of detail, the simplest form that I've seen working is exporting to PSS/e raw (v33 for instance) and import into GridCal.

As for the diagrams, I'm afraid you will have to do some PowerFactory digging to get the coordinates of the elements.

If I were you, I'd make some python script to get traverse the PowerFactory objects and create the GridCal objects (diagrams included)

I don't have a PowerFactory license that I can use, but I can help you understanding things from the GridCal side.

BR,
Santiago

SanPen, 2024-10-01:
@yns0671 How is it going?

yns0671, 2024-10-23:
I could not work on this for a while. Now, I realised that i cant export to pss/e due to our licence. I will try ro find out another way. Thanks for your support.
Greetings from Gökhan Tosun, my manager at MRC.

SanPen, 2024-10-23:
You could try the power factory text file export. GridCal may be able to do something with that file.

Greetings to Gökhan too!

yns0671, 2024-11-04:
I obtain line coordinates from digsilent and use them in order to a create line in gridcal by this code:
It creates two bus on top of each other at "All Bus Branch". What i want is to see this line on map using its coordinates. Or, alternatively, it could be good if i see this bus not on the same location on the line diagram. how can i determine their location on the diagram? This code just add the coorinates as attribute of the line

# take coordinates
terminal_i_coords = {
    'lat': terminal_i.GetAttribute('GPSlat'),
    'lon': terminal_i.GetAttribute('GPSlon')
}
terminal_j_coords = {
    'lat': terminal_j.GetAttribute('GPSlat'),
    'lon': terminal_j.GetAttribute('GPSlon')
}

# create GridCal MultiCircuit 
grid = gce.MultiCircuit()

# create terminals
bus1 = Bus("Bus1", latitude=terminal_i_coords['lat'], longitude=terminal_i_coords['lon'])
grid.add_bus(bus1)

bus2 = Bus("Bus2", latitude=terminal_j_coords['lat'], longitude=terminal_j_coords['lon'])
grid.add_bus(bus2)

# add branch
branch = gce.Branch(bus1, bus2, name=line_name, r=0.05, x=0.11, b=0.02)  # Örnek direnç ve reaktans değerleri
grid.add_branch(branch)

SanPen, 2024-11-04:
could you share a simple gridcal file for me to see the issue?

yns0671, 2024-11-04:
here is an example code:

from GridCalEngine.Devices import Bus  # Substation sınıfını ekledik

import GridCalEngine.api as gce

# Hattın bilgilerini al
line_name = "172845235_WCR_OVERHEAD_ACSR_150MM2_33 KV"  # İlgili hattın ismini belirtiyoruz
line = None

terminal_i_coords = {'lat': 13.4527, 'lon': -16.5780}  # Bus1 
terminal_j_coords = {'lat': 13.4530, 'lon': -16.5785}  # Bus2 

grid = gce.MultiCircuit()

bus1 = Bus("Bus1", latitude=terminal_i_coords['lat'], longitude=terminal_i_coords['lon'])
bus1.Vnom = 110 
bus1.Vmax = 115  
bus1.Vmin = 105 
grid.add_bus(bus1)

bus2 = Bus("Bus2", latitude=terminal_j_coords['lat'], longitude=terminal_j_coords['lon'])
bus2.Vnom = 110 
bus2.Vmax = 115  
bus2.Vmin = 105  
grid.add_bus(bus2)

# Hattı ekle
branch = gce.Branch(bus1, bus2, name=line_name, r=0.05, x=0.11, b=0.02)  # Örnek direnç ve reaktans değerleri
grid.add_branch(branch)

# GridCal dosyasını kaydet
gridcal_file_path = r"C:\Users\path)

print(f"{line_name} created {gridcal_file_path} saved")

SanPen, 2024-11-04:
I think I understand your problem. In a schematic diagram the latitude and longitude are useless, because it is a 2D cartesian representation, hence you need to set the `x`and `y`properties. For that, in the model you can call `grid.fill_xy_from_lat_lon(destructive=True)` That is also available in the user interface in the model menu.

Then, you can click on the `+`or `- ` buttons to make the buses closer or further apart.

closed 2024-11-11 by yns0671