SanPen/VeraGrid #315: Tutorial Bug

MarcoGalici, 2024-10-16:
**5-node grid creation script error**
When running the tutorial I am not able to import the package. 
Running:

 ```
from GridCal.Engine import *
```

I am not able to import the Engine. However, running:

 ```
import GridCalEngine.api as gce
 ```
It works. 

**Desktop (please complete the following information):**
 - OS: [Windows 10]
 - Python Version [3.9]

**Additional context**
Maybe I installed the package in the wrong way, but I strictly followed the tutorial so I am not sure. Thank you for your help anyway 👍🏻

SanPen, 2024-10-16:
Totally agree, could you please tell me where did you see that?

MarcoGalici, 2024-10-17:
Yes sure. I have seen the problem following the tutorial in this webpage [link](https://gridcal.readthedocs.io/en/latest/rst_source/tutorials/five_node_grid.html). 
I have to add that I have found another problem but this regard the python version. I am using python 3.9 but this is my fault since you specify that it is suggested to use the latest python version. As a matter of fact from the problem that I see the issue has been solved from python 3.10  :sweat_smile:

SanPen, 2024-10-18:
Fixed in the docs. 
Yes, since a couple of versions GridCal is for python >= 3.10 to ease development and dependency handling.

closed 2024-10-18 by SanPen