SanPen/VeraGrid #318: Support for Defining a Transformer with Neutral Connection

MarcoGalici, 2024-10-30:
Hello,

I’m working on configuring a transformer with a neutral connection, and I noticed that only three types of connections are available: 

1.  G (grounded star),
2. S (ungrounded star), and
3. D (delta).

However, it seems that a four-wire connection (neutral connection) is not available.

I came across this limitation while following the Distribution Grid tutorial ([link](https://gridcal.readthedocs.io/en/latest/rst_source/tutorials/distribution_grid_standalone.html#)), and I wanted to check if there is a way to set up a transformer with a four-wire (neutral) connection that I may have missed.

Questions:
Is there a way to define a four-wire (neutral) connection for transformers?
If not, is there a recommended workaround for simulating a transformer with a neutral connection?
Are there any plans to support this feature in the future?

Thank you for your help!

SanPen, 2024-10-30:
Hi, As far as I understand, grounded star is what you're asking for.

MarcoGalici, 2024-10-30:
Ok thank you :smiley:. 
To clarify, if I have a 4 wire system, and I have to add a transformer YnD, should I use GD as connection? This means that the software automatically understand that the 4th wire is the neutral and the transformer should be connected to the ground and to the neutral wire, am I right? Thank you for your help.

SanPen, 2024-10-30:
that's it. I think using `G` might not be the greatest nomenclature...maybe I'll change it to 'Y' and 'Yn'

MarcoGalici, 2024-10-30:
Great, thank you very much 😄. I will try it immediately.

closed 2024-10-30 by MarcoGalici