SanPen/VeraGrid #322: Map Diagram Issue: Substation Circles

yns0671, 2024-11-11:
**Is your feature request related to a problem? Please describe.**
Recently, I noticed a regression in the map diagram rendering in GridCal. Previously, the map diagram was clearer and easier to interpret. However, after the latest updates, adding substations results in the drawing of oversized circles, making the diagram less readable and cluttered.

**Describe the solution you'd like**
As before, whe should be able to add substations as lines similar to the ones in the branch diagram. 

**Additional context**

[image]

SanPen, 2024-11-11:
Hi, In the diagram settings you should be able to adjust the nodes sizes by changing the `node min size` parameter. That is going to be linked to the diagram after that. Let me know if it works for you.

yns0671, 2024-11-13:
it worked. thanks

closed 2024-11-13 by yns0671