SanPen/VeraGrid #324: Separate but Interconnected Diagrams for Power Network Sections

yns0671, 2024-11-13:
Is your feature request related to a problem? Please describe.
I'm looking for a way to separate different parts of the power network, such as distribution and transmission, into different diagrams. However, these diagrams should still be connected or linked, as they are part of the same network. It can be challenging to analyze or visualize complex power systems without dividing them into different sections while maintaining interconnections.

Describe the solution you'd like
I would like to have the ability to create separate diagrams for different network sections (e.g., transmission and distribution) and link them so that they act as part of a single network. This would allow each section to be displayed on its own diagram, but any changes or data affecting one diagram would also reflect in the others where they are linked.

SanPen, 2024-11-13:
Hi! This is exactly how it works. 

You can create new diagrams in a variety of ways: 
- diagrams->new diagram from selection
- On the database tables, select some device (i.e a bus) right click, new vecinity diagram

When simulating, all the elements in the database regardless of the diagram are simulated together.

Let me know if you manage.

yns0671, 2024-11-13:
Is there a way to do this on a script. for example i have created one part, to that file i will add the second part with code as a new diagram. or maybe in a single code i will create two grids using grid = gce.MultiCircuit() (gce is Gridcalengine)

SanPen, 2024-11-13:
You can easily do that by adding a circuit in the GUI with "import circuit"

You can generate a schematic diagram with the function `generate_schematic_diagram` located from `from GridCal.Gui.Diagrams.SchematicWidget.schematic_widget import generate_schematic_diagram` and then, I suppose that you need to add the diagram to the circuit with `add_diagram`.

```python
from GridCal.Gui.Diagrams.SchematicWidget.schematic_widget import generate_schematic_diagram

# load the grid or whatever....

diagram = generate_schematic_diagram(buses=circuit.get_buses(),
                                                     busbars=circuit.get_bus_bars(),
                                                     connecivity_nodes=circuit.get_connectivity_nodes(),
                                                     lines=circuit.get_lines(),
                                                     dc_lines=circuit.get_dc_lines(),
                                                     transformers2w=circuit.get_transformers2w(),
                                                     transformers3w=circuit.get_transformers3w(),
                                                     windings=circuit.get_windings(),
                                                     hvdc_lines=circuit.get_hvdc(),
                                                     vsc_devices=circuit.get_vsc(),
                                                     upfc_devices=circuit.get_upfc(),
                                                     series_reactances=circuit.get_series_reactances(),
                                                     switches=circuit.get_switches(),
                                                     fluid_nodes=circuit.get_fluid_nodes(),
                                                     fluid_paths=circuit.get_fluid_paths(),
                                                     explode_factor=1.0,
                                                     prog_func=None,
                                                     text_func=None)

circuit.add_diagram(diagram)
```