SanPen/VeraGrid #325: Crash at large grids

yns0671, 2024-11-18:
**Describe the bug**
We impported a 3000 bus system in order to try but the app is unsuable right now. Is it about graphic setings or something else. Maybe Gridcal is not for such large grids or we do something wrong. Can we fix that?

[image]

SanPen, 2024-11-18:
Hi!

Generally, the issue with the graphics happens if you have many buses too close together. That is a Qt thing.

You have a couple of options:
- Don't generate a 3000 bus diagram, and generate smaller ones.
- Don't generate a schematic diagram and use a map diagram instead to visualize all the grid.
- Separate the nodes sufficiently so that they are displayed individually (plus button)

Let me know if it works

yns0671, 2024-11-25:
it works better when i divide the grid into pieces and add circuits. thanks

closed 2024-11-25 by yns0671