SanPen/VeraGrid #328: Add AC-PTDF to Linear Analysis

leuchtum, 2024-12-11:
**Is your feature request related to a problem? Please describe.**
The current implementation of `‎LinearAnalysisDriver` exposes the PTDF (mathematically based on the DC) as well as the LODF. I'm suggesting to expose the AC-PTDF as well.

**Describe the solution you'd like**
The 'hard' part ist already done, as the method [`make_acptdf`already exist in the source code](https://github.com/SanPen/GridCal/blob/43263a7a5235424778a73837ede8fc8e9f417722/src/GridCalEngine/Simulations/LinearFactors/linear_analysis.py#L81) (Although it's never used and untested). I'm suggesting to add the AC-PTDF to `‎LinearAnalysisDriver`.

**Describe alternatives you've considered**
Currently if you want to use the ACPTDF, you need to use the untested function and "hack" it to a DataFrame representation.

@ Maintainers: Is there a reason, why this haven't been done already? Do I miss something obvious? I'm happy to make an PR to add the AC-PTDF, but I need some guidance before I get started. Also, in the docs you mention the VTDF, but there is no implementation.

SanPen, 2024-12-11:
Hi Daniel, 

The main reason is that the B-matrix based PTDF is almost the same as the one coming from the AC-PTDF, so I did not see much benefits there. Also in practice, people trust to an extreme level the B-matrix PTDF, and use linear models because their models are in voltage collapse, and they merely care about load  ~= generation.  For that, the linear analysis is great. It has been empirically proven to be even better than the linear power flow since you can plainly curtail nonsense values in the PTDF and LODF yielding kind-of-ok flows for odels that are in blackout in practice. Not, to say those modelling practices should not be punished with jail time, but it is what it is.

Rant aside, the simple approach would be to include some new field in the linear options and an if statement when calling the PTDF. 

About the VTDF, long ago I produced such a thing, but I felt it didn't add much. To me the king linear method is what in the power flow options is called "linear ac" which approximates voltage modules and angles. From that you can get the flows. A factorization + a matrix multiplication is in my view a far better thing than computing PTDF-like matrices that crash and burn when the grid is radial.

If you'd like to chat about your use case, you can find me on [discord](https://discord.com/invite/dzxctaNbvu).

SanPen, 2026-09-25:
Finally done at 6.7.1.

The Linear Analysis class allows for the option to use the Jacobian based PTDF. So does the GUI options to run linear analysis.

closed 2026-09-25 by SanPen