SanPen/VeraGrid #337: Issue with reading PSSE transformers

jahanbani, 2025-02-17:
This issue is more of a resiliency of the code issue rather than a bug. I am working with real network data and there are situations that the following happens:

[image]

The code throws an error on line 421 of `raw_to_gridcal.py`
            `elif psse_elm.VMA2 != 0:`
In reality, this line of code should have not been compiled. However, since there is a data entry issue (highlighted yellow in the picture), the code eaches line 421 and throws `TypeError: '>' not supported between instances of 'NoneType' and 'int'`

This is more of a code resiliency which would help in adoptation of the code by industry.

SanPen, 2025-02-18:
Hi, 

It would be super helpful if you could attach a simple .raw file just with the transformers that cause trouble. This way I can create testing for those.

SanPen, 2025-07-22:
I created the `test_issue_337` in `tests/test_load_all_grids.py` and it passes.

@jahanbani  Can you confirm it works for the case you created?

jahanbani, 2025-08-17:
@SanPen 
I will create a small test system with these issue and will let you know.
Meanwhile, I have a question regarding Phase Angle Regulators (PARs). Is there a way to know which transformers are PARs?
It seems that all transformers are converted to fixed tap; see raw_to_grical.py Line 461.
Thank you.

SanPen, 2025-08-17:
Hi, you can determine that by checking the `tap_phase_control_mode` property on the transformers.

The line you mention, sets the active power control at the from side if COD1 > 0, or to fixed otherwise.
```
tap_phase_control_mode = TapPhaseControl.Pf if psse_elm.COD1 > 0 else TapPhaseControl.fixed
```