SanPen/VeraGrid #341: ISO PSS/e model import issues

stuttgart917, 2025-02-28:
I ran into a psse_circuit.py parsing error when trying to import an ISO's recent commercial model.  I tried a second time but with a state estimator file and it failed with a parsing error in switched_shunt.py.  Below are the dumps:

Commercial model (PSS(tm)E-30 RAW created      Thu, Feb 20 2025 14:03)
Error calling Python override of QThread::run(): Traceback (most recent call last):
  File " \gridcal_env\Lib\site-packages\GridCal\Session\file_handler.py", line 109, in run
    self.circuit = file_handler.open(text_func=self.progress_text.emit,
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File " \gridcal_env\Lib\site-packages\GridCalEngine\IO\file_handler.py", line 383, in open
    pss_grid = read_raw(self.file_name,
               ^^^^^^^^^^^^^^^^^^^^^^^^
  File " \gridcal_env\Lib\site-packages\GridCalEngine\IO\raw\raw_parser_writer.py", line 277, in read_raw
    grid.parse(sections_dict['info'][0])
  File " \gridcal_env\Lib\site-packages\GridCalEngine\IO\raw\devices\psse_circuit.py", line 117, in parse
    self.IC, self.SBASE, self.REV, self.XFRRAT, self.NXFRAT, self.BASFRQ = data
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: not enough values to unpack (expected 6, got 3)

State estimator model (PSS(R)E 30 RAW created by rawd30  FRI, FEB 14 2025  21:09)
Error calling Python override of QThread::run(): Traceback (most recent call last):
  File "\gridcal_env\Lib\site-packages\GridCal\Session\file_handler.py", line 109, in run
    self.circuit = file_handler.open(text_func=self.progress_text.emit,
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File " \gridcal_env\Lib\site-packages\GridCalEngine\IO\file_handler.py", line 383, in open
    pss_grid = read_raw(self.file_name,
               ^^^^^^^^^^^^^^^^^^^^^^^^
  File " \gridcal_env\Lib\site-packages\GridCalEngine\IO\raw\raw_parser_writer.py", line 416, in read_raw
    obj.parse(data2, version, logger)
  File " \gridcal_env\Lib\site-packages\GridCalEngine\IO\raw\devices\switched_shunt.py", line 209, in parse
    self.I, self.MODSW, self.ADJM, self.STAT, self.VSWHI, self.VSWLO, \
ValueError: not enough values to unpack (expected at least 10, got 8)

 - OS: Windows 11
 - Python 3.12.5
 - GridCal v5.3.14

Looking forward to your feedback.

stuttgart917, 2025-02-28:
Here is the first line of the estimator file (failed in switched_shunt.py with ValueError: not enough values to unpack (expected at least 10, got 8):
0,   100.00, 30, 0, 1,  60.00     / PSS(R)E 30 RAW created by rawd30  FRI, FEB 14 2025  21:09

Here is the first line of the commercial model file (failed in psse_circuit.py with ValueError: not enough values to unpack (expected 6, got 3):
0,100.0,30 / PSS(tm)E-30 RAW created      Thu, Feb 20 2025 14:03 from host

SanPen, 2025-03-01:
could you try the version in the `devel` branch? I've done some modifications there, but I'd like to close this issue before publishing.

stuttgart917, 2025-03-02:
I installed the devel version this morning in a separate venv and got the following error: "'gridcal' is not recognized as an internal or external command, operable program or batch file."  I can't find gridcal in the scripts folder of the dev version like there is in the main branch.  Could be a user error on my side though.

SanPen, 2025-03-02:
try

`python -c "from GridCal.ExecuteGridCal import runGridCal; runGridCal()"`

This is explained in the installation instructions

stuttgart917, 2025-03-02:
Unfortunately, that failed with: ModuleNotFoundError: No module named 'GridCal.ExecuteGridCal'

I created a venv called gridcal_dev_venv then cloned the GridCal devel branch into it.  After activating the venv. I went into the downloaded GridCal top level directory and ran pip install -e .  It successfully installed the module and I see it listed under pip list, but it fails when I call the startup command inside the virtual environment:

python -c "from GridCal.ExecuteGridCal import runGridCal; runGridCal()"

Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'GridCal.ExecuteGridCal'

It has something to do with package location as when I start interactive python inside the venv it fails to find the module GridCal.ExecuteGridCal, but it can find it if I call GridCal.src.GridCal.ExecuteGridCal.  It still fails as other relative import paths are off.

stuttgart917, 2025-03-04:
Ok, got it to run.  

I'm now getting the following error when I load the commercial model:

Error calling Python override of QThread::run(): Traceback (most recent call last):
  File "\gridcal_dev_venv\GridCal\src\GridCal\Session\file_handler.py", line 109, in run
    self.circuit = file_handler.open(text_func=self.progress_text.emit,
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\gridcal_dev_venv\GridCal\src\GridCalEngine\IO\file_handler.py", line 383, in open
    pss_grid = read_raw(self.file_name,
               ^^^^^^^^^^^^^^^^^^^^^^^^
  File "\gridcal_dev_venv\GridCal\src\GridCalEngine\IO\raw\raw_parser_writer.py", line 414, in read_raw
    obj.parse(data2, version, logger)
  File "\gridcal_dev_venv\GridCal\src\GridCalEngine\IO\raw\devices\bus.py", line 153, in parse
    (self.I, self.NAME, self.BASKV, self.IDE, self.GL, self.BL,
ValueError: too many values to unpack (expected 11)

As well as this error when I load the state estimator model:
Error calling Python override of QThread::run(): Traceback (most recent call last):
  File "\gridcal_dev_venv\GridCal\src\GridCal\Session\file_handler.py", line 109, in run
    self.circuit = file_handler.open(text_func=self.progress_text.emit,
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\gridcal_dev_venv\GridCal\src\GridCalEngine\IO\file_handler.py", line 383, in open
    pss_grid = read_raw(self.file_name,
               ^^^^^^^^^^^^^^^^^^^^^^^^
  File "\gridcal_dev_venv\GridCal\src\GridCalEngine\IO\raw\raw_parser_writer.py", line 414, in read_raw
    obj.parse(data2, version, logger)
  File "\gridcal_dev_venv\GridCal\src\GridCalEngine\IO\raw\devices\switched_shunt.py", line 209, in parse
    self.I, self.MODSW, self.ADJM, self.STAT, self.VSWHI, self.VSWLO, \
ValueError: not enough values to unpack (expected at least 10, got 8)

SanPen, 2025-03-05:
I've just updates, please try again.

stuttgart917, 2025-03-06:
I appreciate the iteration.

The State Estimator RAW file loaded successfully!

The Commercial Model failed with the following error:

Error calling Python override of QThread::run(): Traceback (most recent call last):
  File "\gridcal_dev_venv\gridcal\src\GridCal\Session\file_handler.py", line 109, in run
    self.circuit = file_handler.open(text_func=self.progress_text.emit,
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\gridcal_dev_venv\gridcal\src\GridCalEngine\IO\file_handler.py", line 383, in open
    pss_grid = read_raw(self.file_name,
               ^^^^^^^^^^^^^^^^^^^^^^^^
  File "\gridcal_dev_venv\gridcal\src\GridCalEngine\IO\raw\raw_parser_writer.py", line 414, in read_raw
    obj.parse(data2, version, logger)
  File "\gridcal_dev_venv\gridcal\src\GridCalEngine\IO\raw\devices\bus.py", line 159, in parse
    dta[0:n] = data[0]
    ~~~^^^^^
ValueError: could not broadcast input array from shape (12,) into shape (11,)

SanPen, 2025-03-07:
Alright, we're getting closer. Could you please try with the newest changes?

stuttgart917, 2025-03-14:
Here is the latest when opening the commercial model:

Error calling Python override of QThread::run(): Traceback (most recent call last):
  File "\gridcal_dev_venv\gridcal\src\GridCal\Session\file_handler.py", line 109, in run
    self.circuit = file_handler.open(text_func=self.progress_text.emit,
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\gridcal_dev_venv\gridcal\src\GridCalEngine\IO\file_handler.py", line 387, in open
    self.circuit = psse_to_gridcal(
                   ^^^^^^^^^^^^^^^^
  File "\gridcal_dev_venv\gridcal\src\GridCalEngine\IO\raw\raw_to_gridcal.py", line 1014, in psse_to_gridcal
    bus, bus_shunt = get_gridcal_bus(psse_bus=psse_bus,
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "\gridcal_dev_venv\gridcal\src\GridCalEngine\IO\raw\raw_to_gridcal.py", line 74, in get_gridcal_bus
    area=area_dict[psse_bus.AREA],
         ~~~~~~~~~^^^^^^^^^^^^^^^
KeyError: -0.009

SanPen, 2025-03-15:
this is because I made the modifications to read the areas blindly, and now something that was coming in the area object is being stored in the wrong field. Could you paste here the first rows of the area part of the file?

SanPen, 2025-03-17:
The line reading that data seems correct:

```python
(self.I, self.NAME, self.BASKV, self.IDE, self.GL, self.BL,
  self.AREA, self.ZONE, self.VM, self.VA, self.OWNER) = self.extend_or_curtail(data[0], 11)
```
In this example data, -0.009 corresponds to Va (voltage angle), so I don't know how it ends up being the area...

Could you perform a debug yourself? otherwise I'd ask you for the file to see what's the issue.

commit f52ca31f (2025-03-01): Hopefully fixed #341
closed 2025-03-31 by SanPen