SanPen/VeraGrid #350: MOO_MVRSM minimizer starting point

lbliek, 2025-03-17:
As discussed with Santiago, for MOO_MVRSM it is important to start not just from the previous point, but from a Pareto optimal point. The following code from MOO_MVRSM might help with this (untested with GridCal, but adapted from MOO_MVRSM). Please compare with 
https://github.com/SanPen/GridCal/blob/master/src/GridCalEngine/Utils/NumericalMethods/MVRSM_mo_pareto.py
and test for bugs. It could probably be simplified by using the existing code for dominates().
Also, a new scalarization option was added here (augmented Tchebycheff scalarization), but that part is less important than starting from a Pareto optimal point. 

# MIT License
#
# Copyright (c) 2020 Laurens Bliek
# MVRSM uses a piece-wise linear surrogate model for optimization of
# expensive cost functions with mixed-integer variables.
#
# MVRSM_minimize(obj, x0, lb, ub, num_int, max_evals, rand_evals) solves the minimization problem
#
# min f(x)
# st. lb<=x<=ub, the first num_int variables of x are integer
#
# where obj is the objective function, x0 the initial guess,
# lb and ub are the bounds, num_int is the number of integer variables,
# and max_evals is the maximum number of objective evaluations (rand_evals of these
# are random evaluations).
#
# Laurens Bliek, 06-03-2019
#
# Source: https://github.com/lbliek/MVRSM
# Article: Black-box Mixed-Variable Optimization using a Surrogate Model that Satisfies Integer Constraints,
# 		   by Laurens Bliek, Arthur Guijt, Sicco Verwer, Mathijs de Weerdt

import math
import random
import numpy as np
from typing import List, Tuple
from scipy.linalg.blas import dger
from scipy.optimize import minimize
from GridCalEngine.Utils.NumericalMethods.non_dominated_sorting import non_dominated_sorting, dominates
from GridCalEngine.basic_structures import Vec, Mat, IntVec
import timeit

def relu(x):
    """
    The Rectified Linear Unit (ReLU) function.
    :param x: the input and output vector
    """
    return np.maximum(0, x, out=x)

def relu_deriv(x):
    """
    The derivative of the rectified linear unit function,
    defined with `relu_deriv(0) = 0.5`.
    :param x: the input and output vector
    """
    return np.heaviside(x, 0.5, out=x)

class SurrogateModel:
    """
    SurrogateModel
    """

    def __init__(self, n_obj: int, m: int, c: Mat, W: Mat, b: Vec, reg: float, bounds: List[Tuple[float, float]]):
        """
        Container for the surrogate model data, defined as a linear combination of
        `m` basis functions whose weights `c` are to be trained. The basis function
        `Φ_k(x)` is a ReLU with input `z_k(x)`, a linear function with weights `W_{k, ·}ᵀ`
        and bias `b_k`.
        Let `d` be the number of (discrete and continuous) decision variables.
        :param m: the number of basis functions.
        :param c: the basis functions weights (m×1 vector).  --> changes with multiple objectives
        :param W: the `z_k(x)` functions weights (m×d matrix).
        :param b: the `z_k(x)` functions biases (m×1 vector).
        :param reg: the regularization parameter.
        :param bounds: the decision variable bounds. (scipy's bounds)
[... 548 lines cut ...]
                        value += 1 if direction else -1

                    r *= 2

                next_x[j] = value

            # Continuous exploration
            for j in range(num_int, d):

                value = next_x[j]

                while True:  # re-sample while out of bounds.
                    # Choose a variance that scales inversely with the number of decision variables.
                    # Note that Var(aX) = a^2 Var(X) for any random variable.
                    delta = np.random.normal() * (ub[j] - lb[j]) * 0.1 / math.sqrt(d)

                    if lb[j] <= value + delta <= ub[j]:
                        next_x[j] += delta
                        break

    # apply non-dominated sorting
    y_sorted, x_sorted, sorting_indices = non_dominated_sorting(y_values=y_population.copy(),
                                                                x_values=x_population)

    return y_sorted, x_sorted, y_population, x_population

SanPen, 2025-03-17:
Hi!

I had to do several adjustments, most notably the jacobian:

```python
    def g_jac(self, x):  # change this to have the objective index in the argument   def g(self, x, obj_index):
        """
        Evaluates the Jacobian of the model at `x`.
        :param x: the decision variable values.
        """
        phi_prime = self.phi_deriv(x, out=self.scratch)
        # b = np.multiply(self.c[i,:], phi_prime, out=self.scratch)  # use c[obj_index]
        b = np.empty((self.n_obj, self.m))
        for i in range(self.n_obj):
            b[i, :] = self.c[i, :] * phi_prime
        return b @ self.W  # n_obj×d matrix
```

But it seemed to do something:

[image]

[image]

The version is in the branch: 350_new_mvrsm

Best santiago

SanPen, 2025-03-17:
For reference, the NSGA3 results are:

[image]

They are very close

lbliek, 2025-03-18:
Great, thanks a lot for debugging and testing!

closed 2025-03-18 by lbliek

SanPen, 2025-09-15:
if there is a point at 0,0 then I believe those are reserved space that was not used.
What grid file are you using?

lbliek, 2025-09-15:
> if there is a point at 0,0 then I believe those are reserved space that was not used. What grid file are you using?

I am talking about the plot here: https://github.com/SanPen/VeraGrid/issues/350#issuecomment-2730916020

The technical and investment costs are both 0 for at least one of the solutions, which sounds to me like a bug (and NSGA-III did not have this).

lbliek, 2025-09-15:
Ok that's what I thought, thanks!