SanPen/VeraGrid #356: Export grid as CGMES that was created in GridCal

fhjgch, 2025-04-08:
When I try to save a grid as CGMES, GridCal asks for the boundary set.
How can I provide a boundary set, if I created the grid in GridCal itself without importing it from anywhere?
Is there a way to export a grid as CGMES without the boundary set?

SanPen, 2025-04-08:
CGMES demands a boundary set, as incredibly wrong that is...Maybe you can get it from the ENTSO-e transparency platform.

fhjgch, 2025-04-08:
Does that mean I can insert any existing boundary set?

When I use [this one](https://github.com/SanPen/GridCal/blob/master/src/tests/data/grids/CGMES_2_4_15/TestConfigurations_packageCASv2.0/FullGrid/CGMES_v2.4.15_FullGridTestConfiguration_BD_v1.zip) I get the following error:

```
Traceback (most recent call last):
  File "/home/joajak/Documents/repos/cim-libraries-playground/NT-453/example_grid_gridcal_santiago.py", line 78, in <module>
    gce.save_cgmes_file(
  File "/home/joajak/.pyenv/versions/3.12.9/envs/venv/lib/python3.12/site-packages/GridCalEngine/__init__.py", line 90, in save_cgmes_file
    logger += handler.save_cgmes()
              ^^^^^^^^^^^^^^^^^^^^
  File "/home/joajak/.pyenv/versions/3.12.9/envs/venv/lib/python3.12/site-packages/GridCalEngine/IO/file_handler.py", line 638, in save_cgmes
    cgmes_circuit = gridcal_to_cgmes(gc_model=self.circuit,
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/joajak/.pyenv/versions/3.12.9/envs/venv/lib/python3.12/site-packages/GridCalEngine/IO/cim/cgmes/gridcal_to_cgmes.py", line 1890, in gridcal_to_cgmes
    get_cgmes_cn_nodes_from_tp_nodes(gc_model, cgmes_model, logger)
  File "/home/joajak/.pyenv/versions/3.12.9/envs/venv/lib/python3.12/site-packages/GridCalEngine/IO/cim/cgmes/gridcal_to_cgmes.py", line 360, in get_cgmes_cn_nodes_from_tp_nodes
    device=tn.idtag,
           ^^^^^^^^
AttributeError: 'TopologicalNode' object has no attribute 'idtag'
```
running this code:

```
gce.save_cgmes_file(
    grid=grid9,
    filename="IEEE9.zip",
    cgmes_boundary_set_path="/home/joajak/Documents/repos/GridCal/src/tests/data/grids/CGMES_2_4_15/TestConfigurations_packageCASv2.0/FullGrid/CGMES_v2.4.15_FullGridTestConfiguration_BD_v1.zip",
    cgmes_version=gce.CGMESVersions.v2_4_15
)
```

SanPen, 2025-04-08:
if you update to 2.3.29 it should be fixed;

```python
logger = gce.save_cgmes_file(
    grid=grid9,
    filename="IEEE9.zip",
    cgmes_boundary_set_path="../../tests/data/grids/CGMES_2_4_15/TestConfigurations_packageCASv2.0/FullGrid/CGMES_v2.4.15_FullGridTestConfiguration_BD_v1.zip",
    cgmes_version=gce.CGMESVersions.v2_4_15,
    pf_results=power_flow.results
)

if logger.has_logs():
    logger.print()
    logger.to_xlsx("ieee9_cgmes_logs.xlsx")
```

as a side note, there are some things that the current cgmes implementation needs to change (and the whole industry along with it...) and that is the ConnectivityNodes vs TopologicalNode thing.  See [this](https://gridcal.readthedocs.io/en/latest/rst_source/theory/topology_processing.html)

fhjgch, 2025-04-08:
That worked. Thank you.
Interesting read!