SanPen/VeraGrid #402: An error occurred while using SQLite

lcxc2782, 2025-07-09:
After saving the project file as SQLite, opening the project file displays: Execution failed on sql 'select * from connectivity node': no such table: connectivity .  I don't know if it's my operational error or if there's a bug in the program

**Desktop (please complete the following information):**
 - OS: win11
 - Version 5.3

SanPen, 2025-07-09:
Hi, could you share what did you change? in general:
- pack_unpack is a quite critical piece and changes may alter other parts
- the connectivity nodes will disappear because of this: https://gridcal.readthedocs.io/en/latest/rst_source/theory/topology_processing.html

lcxc2782, 2025-07-09:
> 你好，你能分享一下你做了哪些改变吗？总的来说：
> 
> * pack_unpack 是一个非常关键的部分，修改它可能会改变其他部分
> * 由于这个原因，连接节点很快就会消失：https://gridcal.readthedocs.io/en/latest/rst_source/theory/topology_processing.html

line 57 in pack_unpack.py , like this：  'connectivity node': dev.ConnectivityNode(),   ->   'connectivity': dev.ConnectivityNode(),

lcxc2782, 2025-07-09:
Thank for your reply. If you see this, I think I have some ideas. For example, on the server side, we could put it in Postgres, so we could connect online, right? It seems simpler than creating a service, doesn't it? Additionally, the data exchange in this mode will become more streamlined; at least I have already imported over 100 substation locations from my QGIS (also in Postgres) at once using SQLite.

SanPen, 2025-07-09:
> > 你好，你能分享一下你做了哪些改变吗？总的来说：
> > 
> > * pack_unpack 是一个非常关键的部分，修改它可能会改变其他部分
> > * 由于这个原因，连接节点很快就会消失：https://gridcal.readthedocs.io/en/latest/rst_source/theory/topology_processing.html
> 
> 1. line 57 in pack_unpack.py , like this：  'connectivity node': dev.ConnectivityNode(),   ->   'connectivity': dev.ConnectivityNode(),
> 2. rename the table 'connectivity node' to 'connectivity' in the sqlite.
>    when 2 step done,it's worked.

Changing that line will break many other processes, instead in the sqlite import export we can remove spaces in table names

SanPen, 2025-07-09:
> Thank for your reply. If you see this, I think I have some ideas. For example, on the server side, we could put it in Postgres, so we could connect online, right? It seems simpler than creating a [...]

Not sure what you need, but in general SQL is not performant enough for large power systems. I know this from experience at a large project where we wanted to have a Postgress backend for storage and failed miserably trying to handle large grids with time series.

Another reality is that power systems data does not conform so well with tabular data, it looks like it will, but then you have edge cases that are easily handled by a json file and are a nightmare with pure relational tables.

SanPen, 2025-07-22:
Since connectivity nodes are removed in the next release this shouldn't be a problem.